riscv-collab/riscv-gnu-toolchain #696: syntax error occurs during configure: creating ./config.status

echowwww, 2020-08-18:
I'm building it on my windows machine with MSYS2 + MINGW64.

When runs $ ./configure --prefix=/opt/riscv, I got the following error message:

configure: creating ./config.status
./config.status: line 417: syntax error near unexpected token `)'
./config.status: line 417: `    *\'*) ac_optarg=`$as_echo "$ac_optarg" | sed "s/'/'\\\\\\\\''/g"` ;;'

I also found some error message in config.log, but i'm not sure whether it's related. 

configure:2172: $? = 0
configure:2161: gcc -v >&5
Using built-in specs.
COLLECT_GCC=D:\msys64\mingw64\bin\gcc.exe
COLLECT_LTO_WRAPPER=D:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/10.2.0/lto-wrapper.exe
Target: x86_64-w64-mingw32
Configured with: ../gcc-10.2.0/configure --prefix=/mingw64 --with-local-prefix=/mingw64/local --build=x86_64-w64-mingw32 --host=x86_64-w64-mingw32 --target=x86_64-w64-mingw32 --with-native-system-header-dir=/mingw64/x86_64-w64-mingw32/include --libexecdir=/mingw64/lib --enable-bootstrap --with-arch=x86-64 --with-tune=generic --enable-languages=c,lto,c++,fortran,ada,objc,obj-c++ --enable-shared --enable-static --enable-libatomic --enable-threads=posix --enable-graphite --enable-fully-dynamic-string --enable-libstdcxx-filesystem-ts=yes --enable-libstdcxx-time=yes --disable-libstdcxx-pch --disable-libstdcxx-debug --disable-isl-version-check --enable-lto --enable-libgomp --disable-multilib --enable-checking=release --disable-rpath --disable-win32-registry --disable-nls --disable-werror --disable-symvers --disable-plugin --with-libiconv --with-system-zlib --with-gmp=/mingw64 --with-mpfr=/mingw64 --with-mpc=/mingw64 --with-isl=/mingw64 --with-pkgversion='Rev1, Built by MSYS2 project' --with-bugurl=https://github.com/msys2/MINGW-packages/issues --with-gnu-as --with-gnu-ld
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.2.0 (Rev1, Built by MSYS2 project) 
configure:2172: $? = 0
configure:2161: gcc -V >&5
gcc.exe: error: unrecognized command-line option '-V'
gcc.exe: fatal error: no input files
compilation terminated.
configure:2172: $? = 1
configure:2161: gcc -qversion >&5
gcc.exe: error: unrecognized command-line option '-qversion'; did you mean '--version'?
gcc.exe: fatal error: no input files
compilation terminated.

Thanks for any advice

jim-wilson, 2020-08-24:
I would suggest that your shell is broken or your config.status file is broken.  This should work.  The line in the config.status file should look something like this
``` 
*\'*) ac_optarg=`$as_echo "$ac_optarg" | sed "s/'/'\\\\\\\\''/g"` ;;
```

configure is designed to work with any compiler, so it probes the compiler to see what options it will accept, trying every option that works with some supported compiler or compiler version, and then recording which ones worked and ignoring the ones that didn't work.  So the output in the config.log file is normal, and doesn't indicate an error.

lazyparser, 2021-03-04:
Hi @echowwww 

Had you find a solution for this issue? is this issue still relevant?

TommyMurphyTM1234, 2022-09-27:
No update/activity in over a year.

closed 2022-09-27 by TommyMurphyTM1234