riscv-collab/riscv-gnu-toolchain #697: Enabling and disabling some bitmanip instructions

NazerkeT, 2020-08-20:
Hi there,
I am a newbie in computer architecture and for the present, I have finished implementing Zbb, Zbp, Zbs, Zbe groups of Bit Extension on CVA6(Ariane), and heading towards simulating and running them with benchmarks. So, as long as I did not implement the full bitmanip, I am wondering whether should I configure spike and riscv-gnu-toolchain compilers for switching on and off support for some instructions. Could you advise me on this, please?

Thanks,
Best wishes,

Nelson1225, 2020-08-24:
Hi,

In general, if toolchain have supported some architecture extensions, you can choose what extensions, including their additional extensions, you want by setting the `--with-arch` and `--with-abi` configure options of riscv-gnu-toolchain to build the specific toolchain,

riscv-gnu-toolchain/configure --prefix=`pwd`/build-install --with-arch=rv32gcb_zbb_...

But I'm sure the current status of b-ext.  You can give it a try first.

Nelson

TommyMurphyTM1234, 2022-09-27:
Not a RISC-V GCC toolchain issue.

closed 2022-09-27 by TommyMurphyTM1234