riscv-collab/riscv-gnu-toolchain #700: Inline assembly constraint problem with CSR instructions

berri-1, 2020-08-25:
Using the "I" inline assembly constraint together with Zicsr standard extension instructions only allows access to CSR registers in the range 0x000 to 0x7FF.  For instance, it is not possible to write generic code accessing the cycle counter at 0xC00. Compiling:
```C
static inline __attribute__((always_inline))
int read_csr (const int csr_num)
{
    int rv;
    __asm__ __volatile__ ( "csrr %0, %1" : "=r" (rv) : "I" (csr_num) );
    return rv;
}

static inline __attribute__((always_inline))
int read_cycle (void)
{
    int rv;
    __asm__ __volatile__ ( "csrr %0, 0xC00" : "=r" (rv)  );
    return rv;
}

int test (void)
{
    int works = read_csr(0x780);    // ok
    int error = read_csr(0xC00);    // error
    int cycle = read_cycle();       // ok

    return works | error | cycle;
}
```
results in the error:
> test.c: In function 'test':
> test.c:5:5: warning: 'asm' operand 1 probably does not match constraints
>     5 |     __asm__ __volatile__ ( "csrr %0, %1" : "=r" (rv) : "I" (csr_num) );
>       |     ^~~~~~~
> test.c:5:5: error: impossible constraint in 'asm'

Note that writing the constant directly in the assembly code works as intended (function read_cycle). It is only the constraint that is making the problem.

kito-cheng, 2020-08-25:
Constraint `I` mean `An I-type 12-bit signed immediate.` for RISC-V, you need `i`[2] in this scenario.

[1] https://gcc.gnu.org/onlinedocs/gcc/Machine-Constraints.html
[2] https://gcc.gnu.org/onlinedocs/gcc/Simple-Constraints.html#Simple-Constraints

berri-1, 2020-08-25:
I tried your suggestion and can confirm that it works.
Before posting I tried using negative values with the "I" constraint since I noticed the signed immediate comment but that doesn't work either. But that it is possible to use the "i" constraint for RISC-V as well slipped through my mind. 
Thank you for your pointer!

closed 2020-08-25 by berri-1