riscv-collab/riscv-gnu-toolchain #703: How to assemble RISC-V compressed subset(RVC)?

milos1397, 2020-08-27:
Hello to everyone. When I try to assemble some compressed instruction such as C.JAL, I expect to get in binary file file 16-bit instruction but instead I get 32-bit. C.JAL instruction simply expands into normal JAL 32-bit instruction. 
I use this command for assembling: 
riscv64-unknown-linux-gnu-as -c -mabi=ilp32 -march=rv32imac input.s -o output

Does anyone have idea?

jim-wilson, 2020-08-27:
I would guess that you are reading the objdump output wrong.  c.jal always produces a 16-bit instruction if accepted by the assembler.  However, when we disassemble, we use aliases by default for legibility, so you will see a 16-bit encoding followed by jal.  But the 16-bit encoding is proof that you have a 16-bit instruction.  if you don't want the aliases, add -Mno-aliases when using objdump.

This also works the other way.  If you use jal and the assembler sees that it can fit in c.jal, then the assembler will emit c.jal if compressed support is enabled.  So generally, it isn't necessary to use c.jal.  You can just use jal always.

milos1397, 2020-08-27:
Thank you for response. I openned the binary file and saw that the opcode is for 32-bit JAL instruction.

jim-wilson, 2020-08-27:
What do you mean by "opened"?  You didn't give enough info to reproduce.  You only gave an assembler command line, but didn't give the input file, and didn't show the output.  If I assemble c.jal I get a c.jal encoding in the objdump output.  I don't see a problem other than the default use of aliases.

milos1397, 2020-08-27:
Okay, so I have code in the image Number1.
[image]
 I use this command to compile the code:

riscv64-unknown-linux-gnu-as -c -mabi=ilp32 -march=rv32imac as.s -o obo

Then with objdum I disassembly that code and get the instructions in image Number2.
[image]

And as you can see these instructions are 32-bit

When I use objcopy to extract the text section of the object file to see encoding I get this machine code which represents the C.JAL instructions as usual JAL instructions(opcode).
[image]

jim-wilson, 2020-08-27:
The assembler can only use c.jal when the target is in range.  l2 and l4 are undefined symbols with unknown addresses, so the assembler must assume that they might be out of range, and hence it must use jal.  I didn't realize we could convert c.jal to jal.  I thought that would give an error.  Jal will be relaxed by the linker to c.jal if it turns out that the addresses are in range.  But we can't expand instructions at link time, so we can't convert c.jal to jal at link time.  So the assembler needs to emit jal unless we can prove at assembly time that the address is in range.  Anyways, if you use a valid c.jal instruction in assembly code, it will be assembled as c.jal.  Try for instance
	c.jal .+64

You might want to try using -dr when disassembling to see the relocations.

milos1397, 2020-08-28:
Yeah, the problem was that, asssembler converted C.JAL to JAL because it was out of the range. Thank you.

closed 2021-03-04 by lazyparser