riscv-collab/riscv-gnu-toolchain #704: Error in compilation

dhruvguptadg, 2020-08-28:
Hello, 

I am having the following error:

Assembler messages:
Fatal error: invalid -march= option: 'rv32i'
make: *** [crt0.elf] Error 1

How to solve this?

Thank you,
Dhruv Gupta

jim-wilson, 2020-08-28:
You are probably accidentally running the host (x86?) assembler.  Running gcc with --verbose will show what subprograms the gcc driver is calling, and you will probably see /usr/bin/as being used instead of the riscv as.

As a first step, you might want to verify that you have the riscv assembler installed.  There should be a $prefix/$target/bin/as and a $prefix/bin/$target-as.

Did you build the toolchain yourself?  Did you download it?  How exactly you hit this problem depends on where your toolchain came from.

The problem might also be related to how you are running the compiler.  You didn't include the compiler command line so I can't tell.  But there are obscure compiler options and environment variables that could cause this problem, e.g.
  riscv64-uknown-elf-gcc -B/usr/bin tmp.c
will fails this way, but this solution is this case is "don't do that".

lazyparser, 2021-03-04:
@dhruvguptadg Hi, have you fixed it?

TommyMurphyTM1234, 2022-09-27:
No update/activity in over a year so closing this issue

closed 2022-09-27 by TommyMurphyTM1234