riscv-collab/riscv-gnu-toolchain #705: riscv toolchain error

4perros, 2020-09-01:
Hello
I am trying to compile  code for the bitcraze Crazyflie ai deck, namely the face detection example, I am using gap_sdk version 3.4 and the precompiled gap-riscv-toolchain-Ubuntu-18, when I ran make I get the following error 
/usr/lib/gap_riscv_toolchain/bin/…/lib/gcc/riscv32-unknown-elf/7.1.1/…/…/…/…/
riscv32-unknown-elf/bin/ld: cannot find -lgaptools
/usr/lib/gap_riscv_toolchain/bin/…/lib/gcc/riscv32-unknown-elf/7.1.1/…/…/…/…/
riscv32-unknown-elf/bin/ld: cannot find -lgaplib
collect2: error: ld returned 1 exit status.
Could you help me to solve this issue
Thank you
Rodrigo

kito-cheng, 2020-09-01:
The error message show linker can't find gaptools and gaplib, you might need to specify the library path by `-L`.

4perros, 2020-09-01:
hello
thank you for the quick answer. 
Could you show me how to use this -L. I have tried other ways but it doesnt work for me.
thank you
rodrigo

kito-cheng, 2020-09-02:
Find where is libgaptools.a and libgaplib.a and append the path to `-L`, e.g. libgaptools.a located in `/usr/lib/gap_riscv_toolchain/lib/libgaptools.a`, then pass -L/usr/lib/gap_riscv_toolchain/lib/ to gcc.

4perros, 2020-09-06:
Hello
Thank you again for your advice. I tried your suggestion and found those files in the gap-sdk but not in the gap_riscv_toolchain/lib. So I did the following: 
gcc -L/.../.../gap_sdk/install/gap8_v2/lib/gap/gapuino/ -libgaptools.a -libgaplib.a I get an error  /usr/bin/ld cannot find ligaptools
/usr/bin/ld cannot find -libgaplib I have been trying different ways but I keep getting the same error. Could You point me In the right direction or suggest another solution. 
Thank you again
Regards
Rodrigo

lazyparser, 2021-03-04:
@4perros Hi, is this issue got solved?

TommyMurphyTM1234, 2022-09-27:
No update/activity for over a year and it looks like a user build issue rather than an issue with the RISC-V GCC toolchain itself in any case. Closing this issue.

closed 2022-09-27 by TommyMurphyTM1234