riscv-collab/riscv-gnu-toolchain #706: ICE with branch rvv-intrinsics

joy2myself, 2020-09-02:
Hi. @asmorkalov and me met some internal compiler errors when compiling OpneCV with branch rvv-intrinsics.

```
/home/alexander/Projects/riscv/opencv/modules/imgproc/src/resize.cpp:3977:1: internal compiler error: in tree_to_uhwi, at tree.h:4519
/home/alexander/Projects/riscv/opencv/modules/imgproc/src/smooth.dispatch.cpp:751:1: internal compiler error: in tree_to_uhwi, at tree.h:4519
 3977 | }
      | ^
0x1266fd8 tree_to_uhwi(tree_node const*)
        ../../../riscv-gnu-toolchain/riscv-gcc/gcc/tree.h:4519
0x1266fd8 assemble_variable_contents
        ../../../riscv-gnu-toolchain/riscv-gcc/gcc/varasm.c:2133
0x126a567 assemble_variable(tree_node*, int, int, int)
        ../../../riscv-gnu-toolchain/riscv-gcc/gcc/varasm.c:2327
0x126c47a varpool_node::assemble_decl()
        ../../../riscv-gnu-toolchain/riscv-gcc/gcc/varpool.c:587
0xbdc7bc output_in_order
        ../../../riscv-gnu-toolchain/riscv-gcc/gcc/cgraphunit.c:2582
0xbdc7bc symbol_table::compile()
        ../../../riscv-gnu-toolchain/riscv-gcc/gcc/cgraphunit.c:2819
0xbde716 symbol_table::compile()
        ../../../riscv-gnu-toolchain/riscv-gcc/gcc/cgraphunit.c:3005
0xbde716 symbol_table::finalize_compilation_unit()
        ../../../riscv-gnu-toolchain/riscv-gcc/gcc/cgraphunit.c:3002
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
modules/imgproc/CMakeFiles/opencv_imgproc.dir/build.make:730: recipe for target 'modules/imgproc/CMakeFiles/opencv_imgproc.dir/src/resize.cpp.o' failed
make[2]: *** [modules/imgproc/CMakeFiles/opencv_imgproc.dir/src/resize.cpp.o] Error 1
CMakeFiles/Makefile2:2014: recipe for target 'modules/imgproc/CMakeFiles/opencv_imgproc.dir/all' failed
make[1]: *** [modules/imgproc/CMakeFiles/opencv_imgproc.dir/all] Error 2
Makefile:182: recipe for target 'all' failed
make: *** [all] Error 2
```

```
modules/imgproc/CMakeFiles/opencv_imgproc.dir/src/color_hsv.dispatch.cpp.o
during GIMPLE pass: einline
/home/alexander/Projects/riscv/opencv/modules/imgproc/src/color_hsv.dispatch.cpp: In member function 'void cv::hal::cpu_baseline::{anonymous}::HSV2RGB_b::operator()(const uchar*, uchar*, int) const':
/home/alexander/Projects/riscv/opencv/modules/imgproc/src/color_hsv.dispatch.cpp:358:1: internal compiler error: in maybe_canonicalize_mem_ref_addr, at gimple-fold.c:4903
  358 | } // namespace cv
      | ^
0xd1c08a maybe_canonicalize_mem_ref_addr
        ../../../riscv-gnu-toolchain/riscv-gcc/gcc/gimple-fold.c:4903
0xd20f31 fold_stmt_1
        ../../../riscv-gnu-toolchain/riscv-gcc/gcc/gimple-fold.c:4988
0x101cc6a fold_marked_statements
        ../../../riscv-gnu-toolchain/riscv-gcc/gcc/tree-inline.c:5342
0x1029a65 optimize_inline_calls(tree_node*)
        ../../../riscv-gnu-toolchain/riscv-gcc/gcc/tree-inline.c:5448
0x1cebcf6 early_inliner(function*)
        ../../../riscv-gnu-toolchain/riscv-gcc/gcc/ipa-inline.c:2976
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
modules/imgproc/CMakeFiles/opencv_imgproc.dir/build.make:249: recipe for target 'modules/imgproc/CMakeFiles/opencv_imgproc.dir/src/color_hsv.dispatch.cpp.o' failed
make[2]: *** [modules/imgproc/CMakeFiles/opencv_imgproc.dir/src/color_hsv.dispatch.cpp.o] Error 1
CMakeFiles/Makefile2:2034: recipe for target 'modules/imgproc/CMakeFiles/opencv_imgproc.dir/all' failed
make[1]: *** [modules/imgproc/CMakeFiles/opencv_imgproc.dir/all] Error 2
Makefile:182: recipe for target 'all' failed
make: *** [all] Error 2
```

@kito-cheng Hi Kito. Do you have any idea on them?

asmorkalov, 2020-09-02:
@joy2myself I think the second case should be filed as dedicated ticket. Also please provide steps to reproduce the bug.

joy2myself, 2020-09-02:
Steps to compile OpenCV:

```
git clone git@github.com:joy2myself/opencv.git
cd opencv
git checkout rvv
mkdir build && cd build
cmake -DCMAKE_TOOLCHAIN_FILE=../platforms/linux/riscv64-gcc.toolchain.cmake ../
make -k
```

Suppose the toolchain with branch rvv-intrinsic is installed in /opt/RISCV. If not, you may need edit opencv/platforms/linux/riscv64-gcc.toolchain.cmake.

asmorkalov, 2020-09-02:
The second issue disappears in debug build with `-O0`.

kito-cheng, 2020-09-07:
@joy2myself Thanks for report that, most issue seems like cause by using vector type as member data, I've mention in another issue[1], but I know current GCC didn't give and warning or error message about that, I've update `rvv-intrinsic` branch for riscv-gnu-toolchain, updated version will emit error message when you trying to using vector type within struct, class or union, so could you update the toolchain to latest and try again, I believe it will still build fail, but GCC should emit more readable message instead of internal compiler error now.

We'll update those constraint/limitation of vector type into intrinsic RFC document soon.

And we are really appreciate your work, current implementation are still in early stage, the spec still have few gray zones need to clarified, so feel free to feed back your thought and experience in https://github.com/riscv/rvv-intrinsic-doc

https://github.com/riscv/riscv-gnu-toolchain/issues/701#issuecomment-681852703

joy2myself, 2020-09-07:
@kito-cheng  Hi Kito. Thanks for your reply. I updated the toolchain and it seems really the `struct` issue. I will try to implement the in-memory version you suggested in issue https://github.com/riscv/riscv-gnu-toolchain/issues/701.
Thank you for your appreciation of my work. I‘m glad to give more feed backs later.

closed 2020-09-07 by joy2myself