riscv-collab/riscv-gnu-toolchain #707: Add instruction from P-Extension to binutils

MykolaMoshak, 2020-09-04:
I'm trying to add add8 instruction from P-Extension to binutils, but after recompiling the toolchain I am getting an error:
Error: internal: bad RISC-V opcode (mask error): add8 d,s,t

Encoding of the instruction i took from here: https://github.com/riscv/riscv-p-spec/blob/master/P-ext-proposal.adoc#51-add8-simd-8-bit-addition 

This what i did:
1. Add following lines into riscv-binutils-gdb/include/opcode/riscv-ops.h
`#define MATCH_ADD8  0x4800007f`
`#define MASK_ADD8  0xfe00707f`
`DECLARE_INSN(add8, MATCH_ADD8, MASK_ADD8)`

2. Add following line into riscv-binutils-gdb/opcodes/riscv-opc.c:
`{"add8",        0, INSN_CLASS_I,   "d,s,t", MATCH_ADD8, MASK_ADD8, match_opcode, 0 },`

3. Recompile toolchain and call gcc.

How i can fix this error? 
Many thanks.

Nelson1225, 2020-09-07:
First, check the Figure 1.1 in the RISC-V ISA spec, 1.5 Base Instruction-Length Encoding, 

The p-ext spec uses the 80-bit instruction encoding to encode the 32-bit add8.  It seems to conflict with the ISA spec.  If you use assembler's .insn directive to add these instruction, then it should be fine.  But if you want to add them into the assembler directive, then you should modify the riscv_insn_length.

https://github.com/riscv/riscv-binutils-gdb/blob/riscv-binutils-2.35/include/opcode/riscv.h#L30

Nelson1225, 2020-09-07:
Here is the duplicate issue, https://github.com/riscv/riscv-binutils-gdb/issues/231.  This is a binutils issue, I suggest to close the riscv-gnu-toolchain issue, and then update the later discussion in the binutils' side.

MykolaMoshak, 2020-09-29:
Thank you very much for the quick response. I have changed the recommended line and it helped.

closed 2020-09-29 by MykolaMoshak