riscv-collab/riscv-gnu-toolchain #712: continuing "please implement -msoft-float #595"

dsw, 2020-09-12:
(Since the previous issue #595 https://github.com/riscv/riscv-gnu-toolchain/issues/595 was closed, I opened another one.)

Trying your suggestion, and also linking
against dewdrop-berkeley-softfloat-3, I get this message when I try to link:
"can't link hard-float modules with soft-float modules".

jim-wilson, 2020-09-12:
You either need a compiler that emits soft-float by default, or you need a compiler that was built multilib.  You can get a compiler that is soft-float by default by using --with-arch=X and --with-abi=Y configure options specifying the default arch/abi choices that match the ones you are using.  You can get a multilib compiler by using --enable-multilib when configuring.  A multilib compiler will build a collection of arch/abi choices, which currently is about 28 or so I think.  There are thousands of valid arch/abi combinations though, and it isn't practical to build all of them all of the time, so the multilib solution only works if you are using one of those default arch/abi combination.  Otherwise, you should specify --with-arch/--with-abi when configuring using the arch/abi options you want.

dsw, 2020-09-13:
I built the compiler with --enable-multilib and then tried to compile using

kito-cheng, 2020-09-13:
It seems like your musl didn't handle well on soft-float, and the source code seems like different from upstream musl[1] (line number mis-match), so it's hard to know what happened.

[1] https://git.musl-libc.org/cgit/musl/tree/src/fenv/riscv64/fenv.S

dsw, 2020-09-13:
Ah, inline hardware floating point.  Of course that is not going to work if you turn off hardware floating point.

kito-cheng, 2020-10-12:
Seems it's private `musl` issue rather than `riscv-gnu-toolchain`, and long time no further discussion/response, so I close this issue, but feel free to create issue or reopen this issue if you have further question.

closed 2020-10-12 by kito-cheng