riscv-collab/riscv-gnu-toolchain #713: Is it possible to remove the Linker Relaxation for the Call("auipc+jalr") instruction alone?

SaiVK, 2020-09-16:
Hello Everyone
   Is it possible to remove the Linker Relaxation w.r.t **_auipc + jalr_** ==> **_jal_** instruction, but retain all the other linker relaxations?
Can I just comment out the code places where _**R_RISCV_CALL**_ occurs? Will that work?
If yes, is there any downside to it?

Thank you so much
Sai

jim-wilson, 2020-09-16:
Only if you modify the toolchain sources.  Though I suppose you could try editing object files to change R_RISCV_RELAX relocs to R_RISCV_NONE for this sequence.  You still need to do R_RISCV_CALL.  It is the R_RISCV_RELAX that you want to ignore.  You could try to avoid creating R_RISCV_RELAX in the assembler for this sequence.  See md_apply_fix in gas/config/tc-riscv.c.  Or not relax this sequence in the linker.  See _bfd_riscv_relax_call in bfd/elfnn-riscv.c.

SaiVK, 2020-09-16:
Thank you so much @jim-wilson. I will try doing that.

Nelson1225, 2020-09-17:
> You could try to avoid creating R_RISCV_RELAX in the assembler for this sequence. See md_apply_fix in gas/config/tc-riscv.c. Or not relax this sequence in the linker. See _bfd_riscv_relax_call in bfd/elfnn-riscv.c.

Or you can use ".option norelax" to tell assembler that you don't want to generate R_RISCV_RELAX from now, and then use ".option relax" to re-enable the relaxation in some place.  Here is the example, https://github.com/riscv/riscv-newlib/blob/riscv-newlib-3.2.0/libgloss/riscv/crt0.S#L23.

SaiVK, 2020-09-18:
Thank you so much @Nelson1225. I will try doing that.

kito-cheng, 2020-10-12:
I guess you already got answer, so I close this issue, but feel free to reopen or create a new one if you have further question.

closed 2020-10-12 by kito-cheng