riscv-collab/riscv-gnu-toolchain #714: Problem with building "Error: illegal operands `addi sp,sp,-20*(1<<2)-20*FPREGBYTES'"

Dranzey, 2020-09-16:
Hello everyone! Firstly i hope this is right place to post my problem. If it's not please redirect me to another section. I'm trying to build a Cmake based project on target: Sipeed Longan Nano, which is GD32VF103CBT6 MCU with RISC-V 32-bit core of GigaDevice. During configuration process i faced a problem with "entry.S" file, which was connected to the toolchain. I thinks it's some startup code. This are Assembler messages:

Error: illegal operands `addi sp,sp,-20*(1<<2)-20*FPREGBYTES'
Error: unrecognized opcode `fpstore f0,(20*(1<<2)+0*FPREGBYTES)(sp)'
Error: unrecognized opcode `fpstore f1,(20*(1<<2)+1*FPREGBYTES)(sp)'
Error: unrecognized opcode `fpstore f2,(20*(1<<2)+2*FPREGBYTES)(sp)'
Error: unrecognized opcode `fpload f0,(20*(1<<2)+0*FPREGBYTES)(sp)'
Error: unrecognized opcode `fpload f1,(20*(1<<2)+1*FPREGBYTES)(sp)'

aswaterman, 2020-09-16:
Looks like your entry.S depends on some macros that aren't defined.  Those macros aren't provided by the toolchain; they're presumably defined by some header file in the project you're building.  You should seek support from the authors of the project you're building.

Dranzey, 2020-09-16:
Okay, @aswaterman thank's for an advice.

TommyMurphyTM1234, 2022-09-27:
Not an issue.

closed 2022-09-27 by TommyMurphyTM1234