riscv-collab/riscv-gnu-toolchain #716: -march=rv64im -mabi=lp64 --> can't link hard-float modules with soft-float

suhasskrishanmurthy, 2020-09-21:
Hi, 
  I am trying to compile a FreeRtos program with arch rv64im and abi lp64, its throwing error stating can't link hard-float modules with soft-modules. what should I do, in order to resolve this.
This is the command,
/home/ariane/tandem/toolchain/bin/riscv64-unknown-elf-gcc -c -Wall -Wextra -Wshadow -Wpointer-arith -Wbad-function-cast -Wcast-align -Wsign-compare -Waggregate-return -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Wunused -I. -I/home/ariane/tandem/toolchain/include/spike -I./arch -I./conf -I./include -I../../Source/include -I../Common/include -I../../Source/portable/GCC/RISCV -fomit-frame-pointer -fno-strict-aliasing -fno-builtin -D__gracefulExit -march=rv64im -mabi=lp64 -mcmodel=medany 
This is the error:
Linking....
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: arch/boot.o: can't link hard-float modules with soft-float modules
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file arch/boot.o
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: ../../Source/portable/GCC/RISCV/portasm.o: can't link hard-float modules with soft-float modules
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../Source/portable/GCC/RISCV/portasm.o
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: ../../Source/portable/GCC/RISCV/port.o: can't link hard-float modules with soft-float modules
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../Source/portable/GCC/RISCV/port.o
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: ../../Source/croutine.o: can't link hard-float modules with soft-float modules
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../Source/croutine.o
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: ../../Source/list.o: can't link hard-float modules with soft-float modules
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../Source/list.o
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: ../../Source/queue.o: can't link hard-float modules with soft-float modules
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../Source/queue.o
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: ../../Source/tasks.o: can't link hard-float modules with soft-float modules
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../Source/tasks.o
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: ../../Source/timers.o: can't link hard-float modules with soft-float modules
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../Source/timers.o
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: ../../Source/event_groups.o: can't link hard-float modules with soft-float modules
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../Source/event_groups.o
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: ../../Source/portable/MemMang/heap_2.o: can't link hard-float modules with soft-float modules
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../Source/portable/MemMang/heap_2.o
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: ../../Source/string.o: can't link hard-float modules with soft-float modules
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../../Source/string.o
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: arch/syscalls.o: can't link hard-float modules with soft-float modules
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file arch/syscalls.o
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: arch/clib.o: can't link hard-float modules with soft-float modules
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file arch/clib.o
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: main.o: can't link hard-float modules with soft-float modules
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file main.o
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: ../Common/Minimal/blocktim.o: can't link hard-float modules with soft-float modules
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../Common/Minimal/blocktim.o
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: ../Common/Minimal/countsem.o: can't link hard-float modules with soft-float modules
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../Common/Minimal/countsem.o
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: ../Common/Minimal/recmutex.o: can't link hard-float modules with soft-float modules
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file ../Common/Minimal/recmutex.o
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/crt0.o: In function `_start':
(.text+0x0): undefined reference to `__global_pointer$'
(.text+0x8): undefined reference to `_edata'
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/crtbegin.o: In function `deregister_tm_clones':
crtstuff.c:(.text+0x0): relocation truncated to fit: R_RISCV_HI20 against `__TMC_LIST__'
/home/ariane/tandem/toolchain/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: final link failed: Symbol needs debug section which does not exist
collect2: error: ld returned 1 exit status
Makefile:135: recipe for target 'riscv-spike.elf' failed
make: *** [riscv-spike.elf] Error 1

Thanks in advance.

kito-cheng, 2020-09-21:
I would suggest you rebuild a toolchain with `--with-arch=rv64im --with-abi=lp64`.

lazyparser, 2021-03-04:
@suhasskrishanmurthy Hi, is everything going will? Can we close this issue if it had been solved?

closed 2021-03-05 by suhasskrishanmurthy