riscv-collab/riscv-gnu-toolchain #717: Discussion: if possible to add risc-v support for gcc on android

unicornx, 2020-09-22:
hello,

I am trying to port android gcc to support riscv. I see the latest android cross gcc version is 4.9. I wonder if possible to do porting work with this version, because I know current riscv-gnu-toolchain has moved to 10.x.
Any other people has the same plan as me? Welcome to join me :)

kito-cheng, 2020-09-22:
Back port current RISC-V back-end to 4.9 sound unreasonable since the code base it too much different, I think pick up Android related patch to GCC 10 would be easier, because I guess there is only few compiler driver changes like different dynamic linker path.

Or just using clang/llvm directly is another way to do, RISC-V port on LLVM should be mature enough, but I guess there might need some compiler driver related works for clang too.

unicornx, 2020-09-22:
> Back port current RISC-V back-end to 4.9 sound unreasonable since the code base it too much different, I think pick up Android related patch to GCC 10 would be easier, because I guess there is only [...]

Thank you, Kito.  I see in AOSP, gcc and clang/llvm are both used, I'm trying to figure out why google not just use one(clang?). On android gcc build web-page: “GCC that is used by both the Android platfrom and the NDK.”

Anyway, I will try on gcc porting. Also as I know , google use bionic instead of glibc, that should be another impaction on gcc porting work.

jim-wilson, 2020-09-22:
AOSP stopped using gcc a while ago, which is probably why the latest one is gcc-4.9 because they stopped updating it a while ago.  You have to use llvm/clang instead.  Perhaps you are looking at obsolete web pages?  You didn't give any URLs for me to check.

Asking the same question in 3 different places is annoying.  This is a better place than the forums, but not as good as sw-dev as there is no llvm support in riscv-gnu-toolchain.

unicornx, 2020-09-22:
OK，thanks for your reminder, next time I will use sw-dev to ask such questions.

closed 2020-09-22 by unicornx