riscv-collab/riscv-gnu-toolchain #718: Ubuntu 16.04 no longer supported?

lbmeng, 2020-09-22:
It seems the latest riscv-gnu-toolchain does not build on a Ubuntu 16.04 machine.

The build fails at:

```
g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../.././riscv-gcc/gcc -I../.././riscv-gcc/gcc/. -I../.././riscv-gcc/gcc/../include -I../.././riscv-gcc/gcc/../libcpp/include  -I../.././riscv-gcc/gcc/../libdecnumber -I../.././riscv-gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././riscv-gcc/gcc/../libbacktrace   -o optabs-tree.o -MT optabs-tree.o -MMD -MP -MF ./.deps/optabs-tree.TPo ../.././riscv-gcc/gcc/optabs-tree.c
../.././riscv-gcc/gcc/lto-compress.c: In function ‘int lto_normalized_zstd_level()’:
../.././riscv-gcc/gcc/lto-compress.c:120:36: error: ‘ZSTD_maxCLevel’ was not declared in this scope
   else if (level > ZSTD_maxCLevel ())
                                    ^
../.././riscv-gcc/gcc/lto-compress.c: In function ‘void lto_uncompression_zstd(lto_compression_stream*)’:
../.././riscv-gcc/gcc/lto-compress.c:160:74: error: ‘ZSTD_getFrameContentSize’ was not declared in this scope
   unsigned long long const rsize = ZSTD_getFrameContentSize (cursor, size);
                                                                          ^
../.././riscv-gcc/gcc/lto-compress.c:161:16: error: ‘ZSTD_CONTENTSIZE_ERROR’ was not declared in this scope
   if (rsize == ZSTD_CONTENTSIZE_ERROR)
                ^
../.././riscv-gcc/gcc/lto-compress.c:163:21: error: ‘ZSTD_CONTENTSIZE_UNKNOWN’ was not declared in this scope
   else if (rsize == ZSTD_CONTENTSIZE_UNKNOWN)
                     ^
```

GCC in Ubuntu 16.04 has a version of **gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.12)**

jim-wilson, 2020-09-22:
In lto-compress.c I see
#ifdef HAVE_ZSTD_H
#include <zstd.h>
#endif

HAVE_ZSTD_H is defined if /usr/include/zstd.h exists.  Our Ubuntu 16.04 servers do not have this file.  It is part of the libzstd-dev package and I don't have root access.  I have personal machines with Ubuntu 28.04 and Ubuntu 20.04.  I installed the libzstd-dev package on both and it looks OK.  20.04 has v1.4.4 of zstd, and 18.04 has v1.3.3 of zstd.

Either you have a non-standard zstd.h header file, or an older version that doesn't work with gcc.  In the latter case, we need to update the gcc configure tests to check the version of zstd.h not just the existence of it.  But I don't have the ubuntu 16.04 file so I don't know the version.  This needs to be reported and fixed upstream.

You can use --without-zstd when configuring to work around the problem.

lbmeng, 2020-09-23:
The Ubuntu 16.04 /usr/include/zstd.h has the following version:

```
/* *************************************
*  Version
***************************************/
#define ZSTD_VERSION_MAJOR    0    /* for breaking interface changes  */
#define ZSTD_VERSION_MINOR    5    /* for new (non-breaking) interface capabilities */
#define ZSTD_VERSION_RELEASE  1    /* for tweaks, bug-fixes, or development */
```

> You can use --without-zstd when configuring to work around the problem.

I made the following changes, and it can successfully build the riscv-gnu-toolchain on Ubuntu 16.04:
```
diff --git a/Makefile.in b/Makefile.in
index d3c4e92..d562291 100644
--- a/Makefile.in
+++ b/Makefile.in
@@ -276,6 +276,7 @@ stamps/build-gcc-linux-stage1: $(GCC_SRCDIR) stamps/build-binutils-linux \
                --prefix=$(INSTALL_DIR) \
                --with-sysroot=$(SYSROOT) \
                --with-newlib \
+               --without-zstd \
                --without-headers \
                --disable-shared \
                --disable-threads \
@@ -316,6 +317,7 @@ stamps/build-gcc-linux-stage2: $(GCC_SRCDIR) $(addprefix stamps/build-glibc-linu
                --prefix=$(INSTALL_DIR) \
                --with-sysroot=$(SYSROOT) \
                @with_system_zlib@ \
+               --without-zstd \
                --enable-shared \
                --enable-tls \
                --enable-languages=c,c++,fortran \
```

@jim-wilson Will you file a bug report upstream?

kito-cheng, 2020-09-23:
Seems like it's because version of zstd too old, I guess it require at least zstd 1.3 to having both ZSTD_getFrameContentSize[1] and ZSTD_maxCLevel[2].

[1] https://github.com/facebook/zstd/blob/dev/CHANGELOG#L298
[2] https://github.com/facebook/zstd/issues/254

jim-wilson, 2020-09-23:
Thanks for the info.  Filed upstream as
    https://gcc.gnu.org/bugzilla/show_bug.cgi?id=97183

lbmeng, 2020-09-24:
> Thanks for the info. Filed upstream as
> https://gcc.gnu.org/bugzilla/show_bug.cgi?id=97183

Thank you Jim!

closed 2020-09-24 by lbmeng