riscv-collab/riscv-gnu-toolchain #720: asm operand 1 probably doesn't match constraints

lbmeng, 2020-09-24:
Consider the following case:

```
$ cat test.c
static inline __attribute__ ((optimize(2))) long csr_read(int csr)
{
   long result;
    __asm__ ("csrr %0, %1" : "=r"(result) : "I"(csr));
    return result;
}

long csr_read_v(void)
{
    return csr_read(0x100);
}
```

Compile it via:
```
$ riscv64-unknown-linux-gnu-gcc -o test.o -c test.c
test.c: In function 'csr_read':
test.c:4:5: warning: asm operand 1 probably doesn't match constraints
    4 |     __asm__ ("csrr %0, %1" : "=r"(result) : "I"(csr));
      |     ^~~~~~~
test.c:4:5: error: impossible constraint in 'asm'
```

But if removing the "**static**" keyword, the compilation passes.

So it looks the "static" keyword effects how compiler applies to the __attribute__ to a function?

aswaterman, 2020-09-24:
The `"I"` constraint means that the argument must be a constant.  If you remove the `static` keyword, then the function might be called from another translation unit with an unknown value for the `csr` parameter.  Since `csr` isn't a constant, it doesn't meet the asm constraints.

kito-cheng, 2020-09-24:
In this case, you just confused by the inline semantic https://stackoverflow.com/questions/16245521/c99-inline-function-in-c-file/16245669#16245669

`csr_read` never compiled if you removed `static`, you would got same error when you apply `-fgnu89-inline` option or `extern` to `csr_read`.

For the csr read write case, I would suggest you using macro rather than function, otherwise you might got compile fail when compile without optimization (-O0), the reason is CSR require assembly-time constant, which meaning need compiler to propagate the constant value, but compiler only do that when optimization is enabled.
```
#ifdef __GNUC__

#define read_csr(reg) ({ unsigned long __tmp; \
  asm volatile ("csrr %0, " #reg : "=r"(__tmp)); \
  __tmp; })

#define write_csr(reg, val) ({ \
  asm volatile ("csrw " #reg ", %0" :: "rK"(val)); })

#define swap_csr(reg, val) ({ unsigned long __tmp; \
  asm volatile ("csrrw %0, " #reg ", %1" : "=r"(__tmp) : "rK"(val)); \
  __tmp; })

#define set_csr(reg, bit) ({ unsigned long __tmp; \
  asm volatile ("csrrs %0, " #reg ", %1" : "=r"(__tmp) : "rK"(bit)); \
  __tmp; })

#define clear_csr(reg, bit) ({ unsigned long __tmp; \
  asm volatile ("csrrc %0, " #reg ", %1" : "=r"(__tmp) : "rK"(bit)); \
  __tmp; })

#define rdtime() read_csr(time)
#define rdcycle() read_csr(cycle)
#define rdinstret() read_csr(instret)
```

lbmeng, 2020-09-24:
> For the csr read write case, I would suggest you using macro rather than function, otherwise you might got compile fail when compile without optimization (-O0), the reason is CSR require [...]

Ah this explains that with the `static` keyword, the compilation only passes if I pass `-O2` to the compiler.

Indeed using macro rather than function makes it easier for both optimization on and off cases. I did the the following then.

```
$ cat test.c
#define csr_read(reg) \
    ({ long __tmp; \
    asm volatile ("csrr %0, %1" : "=r"(__tmp) : "I"(reg)); \
    __tmp; })

long csr_read_v(void)
{
    return csr_read(0x100);
}
```

Thanks for the pointers!

closed 2020-09-24 by lbmeng

jim-wilson, 2020-09-24:
You can use __builtin_constant_p to detect whether a function or macro arg is a constant or not at compile time.  This is occasionally useful if you want different code for those two cases.