riscv-collab/riscv-gnu-toolchain #723: GCC 9.2 puts some variables in .sdata2 but GCC 10.1 puts the same ones in .srodata

lbmeng, 2020-09-27:
When testing the PolarFire SoC HSS, I found HSS built by GCC 10.1 does not boot while HSS built by GCC 9.2 it can boot.
See https://github.com/polarfire-soc/hart-software-services/issues/3

It turns out different GCC handles the following codes in HSS differently:

```
const struct InitFunction /*@null@*/ globalInitFunctions[] = {
    // Name                  FunctionPointer      Halt   Restart
    { "HSS_Setup_L2Cache",   HSS_Setup_L2Cache,   false, false },
    { "HSS_Init_RWDATA_BSS", HSS_Init_RWDATA_BSS, false, false },
    { "HSS_BoardInit",       HSS_BoardInit,       false, false },
    { "HSS_UARTInit",        HSS_UARTInit,        false, false },
...
};
const size_t spanOfGlobalInitFunctions = mSPAN_OF(globalInitFunctions);
```

GCC 9.2 puts `spanOfGlobalInitFunctions` into .sdata2 section, while GCC 10.1 puts it into .srodata section.
The proposed fix to HSS is https://github.com/polarfire-soc/hart-software-services/pull/6.

Is this GCC behavior change intentional?

I only found a related GCC changes to .srodata section, that was done by @jim-wilson back in 2017:
[PATCH] Add srodata section support to riscv port (https://gcc.gnu.org/legacy-ml/gcc-patches/2017-12/msg00438.html)
But I don't know why such issue was exposed when upgrading GCC 9 to GCC 10 in 2020?

jim-wilson, 2020-09-27:
The use of sdata2 was a bug that broke some programs, though the breakage was subtle so it took us a while to find it, and also a while to fix it given the lack of gcc developers.
https://gcc.gnu.org/pipermail/gcc-patches/2020-May/545639.html

lbmeng, 2020-09-28:
Thanks Jim. After reading [this discussion thread](https://groups.google.com/a/groups.riscv.org/g/sw-dev/c/JV5Jdh4UjVw) mentioned in the GCC patch link you gave, I now understand this is a GCC 9 bug when handling `-fdata-sections`. Indeed HSS build is using `-fdata-sections` option hence it is affected when GCC 10 fixed the bug.

Here is the test case I used:
```
$ riscv64-unknown-linux-gnu-gcc -o test test.c
$ cat test.c
#include <stdio.h>

const int magic = 8;

int main(char argc, char *argv[])
{
        printf("%d\n", magic);
        return 0;
}
```

Testing results:
```
| GCC ver | without "-fdata-sections" | with "-fdata-sections" |
|   9.2.0 |                  .srodata |                .sdata2 |
|  10.1.0 |                  .srodata |               .srodata |
```

@jim-wilson, does this mean RISC-V GCC will never generate `.sdata2` section? What's the problem of using `.sdata2` as the section name? By looking at [this link](https://stackoverflow.com/questions/61319596/what-is-the-significance-of-sbss2-sdata2), it says "_The special section .sdata2 is intended to hold initialized read-only small data that contribute to the program memory image._". Sounds like we are using `.srodata` name to do the same on RISC-V, but why the name matters?

PS: should all these section names be put in the [RISC-V psABI doc](https://github.com/riscv/riscv-elf-psabi-doc/blob/master/riscv-elf.md)?

jim-wilson, 2020-09-28:
Use of sdata2 breaks some programs.  So we had to fix it.  It was only used by accident.  It never sho8uld have been used in the first place.

srodata is a small rodata section accessible from gp.  sdata2 is not accessible from gp.  sdata2 is something different than sdata.

This stuff could go in the psabi is someone writes a patch to do that.

lbmeng, 2020-09-29:
> Use of sdata2 breaks some programs. So we had to fix it. It was only used by accident. It never sho8uld have been used in the first place.
> 
> srodata is a small rodata section accessible from gp. sdata2 is not accessible from gp. sdata2 is something different than sdata.

What is `.sdata2` by definition? I could not find it anywhere on the internet that has a better explanation except the [link](https://stackoverflow.com/questions/61319596/what-is-the-significance-of-sbss2-sdata2) I gave above. Based on the link, it's almost the same as `.srodata`, except the fact that it cannot be accessed via gp.

But if it's only the gp access issue, shouldn't we fix the default linker script to put `.sdata2` in the same section as `.srodata` so that it can be accessed via gp??

Currently we have the following linker scripts:
```
$ riscv64-unknown-linux-gnu-ld --verbose
GNU ld (GNU Binutils) 2.35
  Supported emulations:
   elf64lriscv
   elf64lriscv_lp64f
   elf64lriscv_lp64
   elf32lriscv
   elf32lriscv_ilp32f
   elf32lriscv_ilp32
using internal linker script:
==================================================
/* Script for -z combreloc */
/* Copyright (C) 2014-2020 Free Software Foundation, Inc.
   Copying and distribution of this script, with or without modification,
   are permitted in any medium without royalty provided the copyright
   notice and this notice are preserved.  */
OUTPUT_FORMAT("elf64-littleriscv", "elf64-littleriscv",
              "elf64-littleriscv")
OUTPUT_ARCH(riscv)
...
  .sdata2         :
  {
    *(.sdata2 .sdata2.* .gnu.linkonce.s2.*)
  }
...
  .sdata          :
  {
    __SDATA_BEGIN__ = .;
    *(.srodata.cst16) *(.srodata.cst8) *(.srodata.cst4) *(.srodata.cst2) *(.srodata .srodata.*)
    *(.sdata .sdata.* .gnu.linkonce.s.*)
  }
...
```

If it is confirmed that RISC-V GCC should never generate `.sdata2` section, we should consider removing it in the default linker script, no?

> This stuff could go in the psabi is someone writes a patch to do that.

It's really good if special section like .sdata .srodata usage can be documented clearly in the RISC-V psABI.

jim-wilson, 2020-09-29:
You can find a definition of sdata2 in the PPC EABI.  The PPC EABI reserves a register that points at this address. It is something different from srodata.

I'm sorry that this has inconvenienced you, but use of sdata2 in this case is very wrong, and the bug had to be fixed.  Serious toolchain bugs must be fixed.  It is just a one time change.  It only affects people that have modified linker scripts that incorrectly dropped the srodata section support, so there should be limited impact.  There is no way to paper over the problem.  It was a bug that had to be fixed.  You just need to accept that.

There might be people that want to use sdata2, so it would be wrong to remove it.

I don't think that the psABI is the correct place to document idiosyncracies of GNU ELF support.  The gABI would be more correct.  But we would need a GNU/Linux specific gABI document.  The Linux Foundation used to maintain docs like this, but I am not sure if they still do.
https://refspecs.linuxfoundation.org/

lbmeng, 2020-09-30:
> You can find a definition of sdata2 in the PPC EABI. The PPC EABI reserves a register that points at this address. It is something different from srodata.

So PPC is the only architecture that supports `.sdata2`? Where is `.srodata` defined then? Which archictecture supports `.srodata` which which does not?

> I'm sorry that this has inconvenienced you, but use of sdata2 in this case is very wrong, and the bug had to be fixed. Serious toolchain bugs must be fixed. It is just a one time change. It only [...]

Yes, I understand this is a serious toolchain bug that has to be fixed. It's not a problem for me to accept that :). But I still don't understand all of these bits and hope things are clarified.

> There might be people that want to use sdata2, so it would be wrong to remove it.

Sorry I don't get it. Why people want to use `.sdata2` if RISC-V toolchain never generates that section?? If it's a PPC only section, the `.sdata2` should not exist in the RISC-V default linker script.

> I don't think that the psABI is the correct place to document idiosyncracies of GNU ELF support. The gABI would be more correct. But we would need a GNU/Linux specific gABI document. The Linux [...]

Thank you Jim for the information so far.

jim-wilson, 2020-09-30:
I believe that PPC EABI (but not PPC Linux) is the one that created sdata2 sections in the first place.  It may not be the only one using sdata2.

srodata is the obvious combination of rodata and sdata.  Just grepping ld sources, I see that mep, d30v, m68hc11, m9s12x (?), xgate, and ia64vma also have references to it, though some of those might not actually be using it.

Things get complicated.  Sometimes people create special secctions for one target and it only ever gets used by that target.  Sometimes special sections get used by multiple targets.  Sometimes people create general sections for one target with the idea that other targets may use it later.  Sometimes those sections get used by other targets.  Sometimes they don't.  sdata2 was added as a general section, that is defined by default for all targets that have small data section support.  So removing it from riscv would create extra work, and the possibiliy of problems in the future if someone in the future needs it or wants it.  So even though I think only PPC EABI (but not PPC Linux) uses it, it is handled by the default linker script for all targets with small data support, and it would be unwise to change that.

srodata on the other hand is not a general section, and people have to explicitly add support for it to their target dependent linker script stubs to have it.

If you have a great deal of time available, you could volunteer to do FSF Binutils work and help clean up this mess.

lbmeng, 2020-09-30:
That's a lot of useful information. Thank you very much Jim. I hope someday these mess could be clean up ...

jim-wilson, 2020-09-30:
Fundamentally, I think this is a gcc design bug.  Defining TARGET_ASM_SELECT_SECTION is common.  But through an odd set of circumstances, defining this also sometimes requires defining TARGET_ASM_UNIQUE_SECTION which is not commonly defined.  The connection is non-obvious and not documented.  So we fell into a trap.  I think there are other ports that have similar bugs.  The port maintainers just haven't noticed yet.

if someone had time and inclination to do gcc work, you could try to redesign gcc to prevent this bug from happening, and fix the other suspected targets that have the same problem.

TommyMurphyTM1234, 2022-09-27:
Not a RISC-V GCC toolchain specific issue.

closed 2022-09-27 by TommyMurphyTM1234