riscv-collab/riscv-gnu-toolchain #724: facing issue in installation of riscv gnu toolchain

Prarthi1703, 2020-10-01:
I am installing the riscv gnu compiler toolchain from openHW group github repository. While running the below command I am facing an error with ldu instruction.

./configure --prefix=/opt/riscv
make linux   

It seems like ldu instructions that are RV1281 instructions, which are not supported in the gnu tool chain.

[image]

jim-wilson, 2020-10-01:
We can only help you if you give us enough info to reproduce the problem.  You haven't stated exactly what sources you are using.    I don't know what you mean by OpenHW repo.  Try giving a URL.  Unless you mean you are using openhw hardware which isn't relevant, since only the riscv-gnu-toolchain software matters here.  You haven't stated exactly how you configured the toolchain.  All I know if that this is a rv128i instruction, and we don't have rv128i support, so I'm guessing that you are using modified sources, and you need to talk to whoever modified the sources.

Prarthi1703, 2020-10-01:
Hi Jim,

I am using the link below to install the riscv gnu tool chain.

https://github.com/pulp-platform/pulp-riscv-gnu-toolchain

Thanks & Regards
Prarthi Bhatt

jim-wilson, 2020-10-01:
You will have to ask the openhw/pulp people for help.  This is a heavily modified version of 3 year old FSF GCC sources.  I can't support it.

Prarthi1703, 2020-10-03:
Hi Jim,

Thanks for informing me.

I think I have installed an old version of  riscv gnu toolchain.
Can you please send me the link of the latest version of gnu toolchain.

Thanks & Regards
Prarthi Bhatt

Prarthi1703, 2020-10-03:
Hi,

https://github.com/riscv/riscv-gnu-toolchain
Is the above link the latest version of riscv gnu toolchain ?

Thanks & Regards
Prarthi Bhatt

kito-cheng, 2020-10-12:
> https://github.com/riscv/riscv-gnu-toolchain
> Is the above link the latest version of riscv gnu toolchain ?

The answer dependent what you need, this is vanilla RISC-V GCC without PULP extension, and close this issue since you already got the answer from riscv-sw mailing list.

closed 2020-10-12 by kito-cheng