riscv-collab/riscv-gnu-toolchain #725: Error in installation

srinivasans74, 2020-10-01:
I have followed all the steps to installation.
I have installed the prerequisite too.
I fixed the path too when I run the make command I get the following  error
ake[2]: Entering directory '/home/srinivasan/RISCV/riscv-gnu-toolchain/build-gcc-newlib-stage1/fixincludes'
gcc -c -g -O2 -W -Wall -Wwrite-strings -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition -Wmissing-format-attribute -Wno-overlength-strings -pedantic -Wno-long-long   -DHAVE_CONFIG_H -I. -I../.././riscv-gcc/fixincludes -I../include -I../.././riscv-gcc/fixincludes/../include ../.././riscv-gcc/fixincludes/fixincl.c
gcc -c -g -O2 -W -Wall -Wwrite-strings -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition -Wmissing-format-attribute -Wno-overlength-strings -pedantic -Wno-long-long   -DHAVE_CONFIG_H -I. -I../.././riscv-gcc/fixincludes -I../include -I../.././riscv-gcc/fixincludes/../include ../.././riscv-gcc/fixincludes/fixtests.c
gcc -c -g -O2 -W -Wall -Wwrite-strings -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition -Wmissing-format-attribute -Wno-overlength-strings -pedantic -Wno-long-long   -DHAVE_CONFIG_H -I. -I../.././riscv-gcc/fixincludes -I../include -I../.././riscv-gcc/fixincludes/../include ../.././riscv-gcc/fixincludes/fixfixes.c
gcc -c -g -O2 -W -Wall -Wwrite-strings -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition -Wmissing-format-attribute -Wno-overlength-strings -pedantic -Wno-long-long   -DHAVE_CONFIG_H -I. -I../.././riscv-gcc/fixincludes -I../include -I../.././riscv-gcc/fixincludes/../include ../.././riscv-gcc/fixincludes/server.c
gcc -c -g -O2 -W -Wall -Wwrite-strings -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition -Wmissing-format-attribute -Wno-overlength-strings -pedantic -Wno-long-long   -DHAVE_CONFIG_H -I. -I../.././riscv-gcc/fixincludes -I../include -I../.././riscv-gcc/fixincludes/../include ../.././riscv-gcc/fixincludes/procopen.c
gcc -c -g -O2 -W -Wall -Wwrite-strings -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition -Wmissing-format-attribute -Wno-overlength-strings -pedantic -Wno-long-long   -DHAVE_CONFIG_H -I. -I../.././riscv-gcc/fixincludes -I../include -I../.././riscv-gcc/fixincludes/../include ../.././riscv-gcc/fixincludes/fixlib.c
gcc -c -g -O2 -W -Wall -Wwrite-strings -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition -Wmissing-format-attribute -Wno-overlength-strings -pedantic -Wno-long-long   -DHAVE_CONFIG_H -I. -I../.././riscv-gcc/fixincludes -I../include -I../.././riscv-gcc/fixincludes/../include ../.././riscv-gcc/fixincludes/fixopts.c
gcc -g -O2 -static-libstdc++ -static-libgcc  -o fixincl fixincl.o fixtests.o fixfixes.o server.o procopen.o fixlib.o fixopts.o ../libiberty/libiberty.a
echo timestamp > full-stamp
srcdir="../.././riscv-gcc/fixincludes" /bin/bash ../.././riscv-gcc/fixincludes/mkfixinc.sh riscv64-unknown-elf
sed -e 's/@gcc_version@/10.2.0/' < mkheaders.almost > mkheadersT
mv -f mkheadersT mkheaders
make[2]: Leaving directory '/home/srinivasan/RISCV/riscv-gnu-toolchain/build-gcc-newlib-stage1/fixincludes'
make[2]: Entering directory '/home/srinivasan/RISCV/riscv-gnu-toolchain/build-gcc-newlib-stage1/gcc'
TARGET_CPU_DEFAULT="" \
HEADERS="auto-host.h ansidecl.h" DEFINES="" \
/bin/bash ../.././riscv-gcc/gcc/mkconfig.sh config.h
TARGET_CPU_DEFAULT="" \
HEADERS="options.h insn-constants.h config/elfos.h config/newlib-stdint.h config/riscv/riscv.h config/riscv/elf.h config/initfini-array.h defaults.h" DEFINES="LIBC_GLIBC=1 LIBC_UCLIBC=2 LIBC_BIONIC=3 LIBC_MUSL=4 TARGET_RISCV_ATTRIBUTE=1" \
/bin/bash ../.././riscv-gcc/gcc/mkconfig.sh tm.h
TARGET_CPU_DEFAULT="" \
HEADERS="config/riscv/riscv-protos.h tm-preds.h" DEFINES="" \
/bin/bash ../.././riscv-gcc/gcc/mkconfig.sh tm_p.h
TARGET_CPU_DEFAULT="" \
HEADERS="auto-host.h ansidecl.h" DEFINES="" \
/bin/bash ../.././riscv-gcc/gcc/mkconfig.sh bconfig.h
g++ -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  -DGENERATOR_FILE -fno-PIE -I. -Ibuild -I../.././riscv-gcc/gcc -I../.././riscv-gcc/gcc/build -I../.././riscv-gcc/gcc/../include  -I../.././riscv-gcc/gcc/../libcpp/include  \
	-o build/genmodes.o ../.././riscv-gcc/gcc/genmodes.c
/home/srinivasan/RISCV/riscv/riscv64-unknown-elf/bin/as: unrecognized option '--64'
make[2]: *** [Makefile:2724: build/genmodes.o] Error 1
make[2]: Leaving directory '/home/srinivasan/RISCV/riscv-gnu-toolchain/build-gcc-newlib-stage1/gcc'
make[1]: *** [Makefile:4388: all-gcc] Error 2
make[1]: Leaving directory '/home/srinivasan/RISCV/riscv-gnu-toolchain/build-gcc-newlib-stage1'
make: *** [Makefile:440: stamps/build-gcc-newlib-stage1] Error 2

jim-wilson, 2020-10-01:
You have dot "." on your PATH.  The gcc build procress creates a link to the RISC-V assembler in the build dir so that we can compile RISC-V libraries with the just built RISC-V gcc.  However, if you have dot on your PATH, then this RISC-V assembler will accidentally be used by the native x86_64 compiler, and you get weird errors like this, as the RISC-V as doesn't accept x86_64 assembler options.  The preferred solution is to remove dot from your PATH.  The alternative solution is to put it at the end.

Note that having dot on your PATH is a security risk, as an attacker could put a fake program like ls in a dir that you might cd into, and then when you run ls in that dir they can take control of your account.  This is why the general recommendation is to not put dot on your PATH.  If you want to run a program in the current directory, you should use ./a.out instead of putting dot on your PATH.

Also note that having a PATH that has two semicolons next to each other, or a semicolon at the start or end, is the same as putting dot on your PATH.

srinivasans74, 2020-10-01:
> You have dot "." on your PATH. The gcc build procress creates a link to the RISC-V assembler in the build dir so that we can compile RISC-V libraries with the just built RISC-V gcc. However, if you [...]

srinivasans74, 2020-10-01:
Thanks its solved. How did you find there was an issue in my path variable?

jim-wilson, 2020-10-01:
I've been doing this for over 30 years.  I've seen the same problem enough times that it is obvious to me.

kito-cheng, 2020-10-05:
I assume this issue is resolved.

closed 2020-10-05 by kito-cheng