riscv-collab/riscv-gnu-toolchain #726: Does the GNU MP Library have a RiscV port?

SaiVK, 2020-10-02:
Hello Everyone
   Is there a RiscV port for the GNU MP library? I want to use that in crypto-libraries?
If there is one, can you point me to the build/installation instructions?

Thank you so much
Sai

kito-cheng, 2020-10-05:
this issue is out of scope of `riscv-gnu-toolchian`, but I am in a meeting so I did some try for that, GMP has generic implementation which implement in C, so in theory it support all target without extra porting effort.

There is two scenario  to build GMP:
1. Native build, you want to build that under RISC-V linux: that's just same as how you build on x86 environment. 
2. Cross-build, you want to build that on x86, but run on RISC-V: similar native build but add `--host=riscv64-unknown-elf` or `--host=riscv64-unknown-linux-gnu` to configure script.

I've verified cross build can build successful on my Ubuntu 18.04  machine :)

SaiVK, 2020-10-05:
Apologies for posting this issue over here. Got confused, by looking at **GNU** in GNU MP library.
Thank you so much @kito-cheng. Instead of **_riscv64-unknown-linux-gnu_** can we use **_riscv64-unknown-elf_**. Is that possible?

Thank you so much

kito-cheng, 2020-10-11:
Seems resolved, so I close this issue :)

closed 2020-10-11 by kito-cheng