riscv-collab/riscv-gnu-toolchain #730: riscv gnu toolchain installation error

Prarthi1703, 2020-10-03:
I'm installing the latest version of the riscv gnu tool chain.

./configure --prefix=/opt/riscv --with-arch=rv32g --with-abi=ilp32d
make linux

[image]

jim-wilson, 2020-10-12:
Do you already have a toolchain in /opt/riscv?  The build can fail if there is already one there that was configured differently.  It is best to use a unique prefix for every build.  Otherwise, you should check the relavent config.log file for an error.

lazyparser, 2021-03-04:
@Prarthi1703 Hi, have you fixed it?

TommyMurphyTM1234, 2022-09-27:
No update/activity for over a year so closing this issue.

closed 2022-09-27 by TommyMurphyTM1234