riscv-collab/riscv-gnu-toolchain #731: Facing Makefile version error

Prarthi1703, 2020-10-05:
I am running this below command to build cross-compile.

Running this command and getting an error.

./configure --prefix=/opt/riscv
make linux 

 Referring site : https://github.com/riscv/riscv-gnu-toolchain  
 I am using the GNU Make 3.82 version.

I try yum install make this command but still getting
[image]
[image]

aswaterman, 2020-10-05:
Building glibc requires GNU make 4.0.  It was released 7 years ago, so if executing `yum check-update` before `yum install make` doesn't do the trick, it might be time to upgrade your operating system.

Prarthi1703, 2020-10-05:
Okay.....thanks for updating.

I will try yum check-update before yum install make.

Prarthi1703, 2020-10-05:
I tried try yum check-update before yum install make.

but i giving error.

[image]

Prarthi1703, 2020-10-05:
I tried try yum check-update before yum install make.

but I giving error.

[image]

aswaterman, 2020-10-05:
Sounds like your OS' package list is rather old.  There are various things you could try, including third-party package lists, building make from source, etc.  Or just upgrade your OS.  Unfortunately, we can't provide support for any of these things.

lazyparser, 2021-03-04:
@Prarthi1703 Hi you can just download the source code of the latest stable CMake, build it, install it to a non-root folder, and modify `PATH` env to add the folder into the search list of binaries. Then you will fix this issue.

TommyMurphyTM1234, 2022-09-27:
Not an issue with the RISC-V GCC toolchain.

closed 2022-09-27 by TommyMurphyTM1234