riscv-collab/riscv-gnu-toolchain #733: Relocation truncated to fit: R_RISCV_HI20 in Assembly

Phantom1003, 2020-10-08:
_start is at 0x80000000
```
.global _start
_start:
  lui sp, %hi(_start)           # fail
  lui sp, %pcrel_hi(_start) # success
```
I found that it seems -mcmodel=medany is ued for `cc1`, cause following works
```
// in c, compile with -mcmodel=medany
	extern long _start;
	_start = 0xdeadbeef;

```

Does there any option is uesd to enable high memory space for `as` ? Or we must use %pcrel_hi ?

jim-wilson, 2020-10-12:
For rv64, lui sign extends to 64-bits, so you can use lui to load 0xffffffff80000000 but not 0x0000000080000000.  This is a hardware design issue, not an assembler issue.  For rv32, you can use lui to load 0x80000000.

pcrel_hi will only work correctly with auipc.  pcrel_hi is complicated to use.  See the psabi for details.

Phantom1003, 2020-10-13:
Thanks, I read the paragraphs following https://github.com/riscv/riscv-elf-psabi-doc/blob/master/riscv-elf.md#medium
In fact, lui t0, 0x80000000 works also.
I wonder to know why _start is translated to 0xffffffff80000000 instead of 0x0000000080000000

aswaterman, 2020-10-13:
The address 0x80000000 is out of range for the medlow code model in RV64.  Using `lui` and `%hi` won't work.  For a simple solution, you could try using the `la` macro, e.g. `la reg, symbol`, which will generate the `auipc` and `%pcrel_hi` for you.

Phantom1003, 2020-10-13:
> The address 0x80000000 is out of range for the medlow code model in RV64. Using `lui` and `%hi` won't work. For a simple solution, you could try using the `la` macro, e.g. `la reg, symbol`, which will generate the `auipc` and `%pcrel_hi` for you.

Thanks, I have known how to make it works. But ...
What's difference between lui reg, %hi(label) and lui reg, %hi(cosnt).

jim-wilson, 2020-10-13:
We can resolve %hi(const) at assembly time, but %hi(label) can't be resolved until link time.

Phantom1003, 2020-10-13:
but why ld can't get the right address and need to signed extension...  it sounds weird

jim-wilson, 2020-10-13:
Sign extension is a hardware property.  See the ISA manual.  Sign extending all 32-bit values to 64-bits means we don't need 32-bit compare and branch instructions.  We can do 32-bit compares with 64-bit compare instructions.

Phantom1003, 2020-10-13:
ohh, i get your mind. even i finally got lui t0, 0x800 in asm.
after execution, in t0, the value is ffffffff80000000, which is not as same as my expect. this also explain why we need medium option, right ?
back to the question, i think the problem is the error info is not clear for me to get this point ... 
thanks!

Phantom1003, 2020-10-14:
#103

closed 2020-10-14 by Phantom1003