riscv-collab/riscv-gnu-toolchain #734: Idea: Option to force alignment of compressed instructions

stnolting, 2020-10-10:
Maybe I am on the wrong track, but imagine a `rv32ic` processor that can access instruction memory/cache only on 32-bit boundaries. If the processor does a branch to an unaligned address (let's say `0x00000002`) and detects an uncompressed instruction there, it has to do another instruction fetch to get the second half of that instruction. Obviously, this slows down the performance.

Is there any option available (or planned) to force an "alignment" of compressed instructions?
So that any odd sequence of compressed instructions gets "padded" with a compressed NOP at the end/at the beginning and
all uncompressed instructions will always be 32-bit aligned?

Basically, something like `-mstrict-align` but for instruction fetches?

Or is this case too special? :thinking:

kito-cheng, 2020-10-12:
I think `-falign-jumps=4 -falign-functions=4  -falign-labels=4 -falign-loops=4` is what you want, and here[1] is GCC document for those options :)

[1] https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html#Optimize-Options#index-falign-functions

stnolting, 2020-10-12:
That was exactly what I was looking for.
Thank you very much! :smile:

closed 2020-10-12 by stnolting