riscv-collab/riscv-gnu-toolchain #736: A bug of qemu "SIOCGSTAMP"

pz9115, 2020-10-13:
I just build all toolchains， test all built use 

```
make check
```

and a error occurs :

[image]

How could I fix it?

kito-cheng, 2020-10-13:
I saw it might fixed on upstream[1-2], could you checkout to qemu `v4.2.1` manually and try to again?

[1] https://patchwork.kernel.org/patch/10998911/
[2] https://patchwork.kernel.org/patch/11048013/

pz9115, 2020-10-13:
> I saw it might fixed on upstream[1-2], could you checkout to qemu `v4.2.1` manually and try to again?
> 
> [1] https://patchwork.kernel.org/patch/10998911/
> [2] https://patchwork.kernel.org/patch/11048013/

It fix after update the qemu repositories,thks!

closed 2020-10-13 by pz9115