riscv-collab/riscv-gnu-toolchain #737: Unnecessary andi instruction when loading volatile bool

kjetilos, 2020-10-13:
When loading a volatile bool value I see that the compiler adds unnecessary andi instruction. Here is reproducer:

```C
#include <stdbool.h>

extern volatile bool active;

int foo(void)
{
  if (!active) {
    return 42;
  } else {
    return -42;
  }
}
```

And the code generated looks like this:
```
foo:
        lui     a5,%hi(active)
        lbu     a5,%lo(active)(a5)
        li      a0,42
        andi    a5,a5,0xff
        beq     a5,zero,.L1
        li      a0,-42
.L1:
        ret
```
The andi instruction seems to be unnecessary since lbu is zero extending the byte.

aswaterman, 2020-10-13:
This isn't a RISC-V-specific issue; other architectures' GCC ports have similar idiosyncrasies, including x86-64.

kjetilos, 2020-10-13:
I see, is this issue reported somewhere else or should I report it to some other repo/mailinglist?

jim-wilson, 2020-10-13:
You can report issues at gcc.gnu.org

Some optimizations must be deliberately disabled for volatiles to avoid breaking the semantics of volatile, and that makes it hard to generate optimal code for volatile accesses.

This problem has been reported before, but no one has volunteered yet to try to fix it.

kjetilos, 2020-10-14:
Thanks Jim, I will report it to gcc.gnu.org as well.

jim-wilson, 2020-10-15:
Was reported upstream
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=97417

closed 2020-10-15 by jim-wilson