riscv-collab/riscv-gnu-toolchain #738: Relocation trunacted to fit error R_RISCV_JAL while compiling kernel

atishp04, 2020-10-14:
Hi,
I am making some changes to improve kernel protection. I tried to move __init.text below __text both being in a different 2MB sections. This is done to apply different permissions to these areas. Kernel uses 2MB(PMD) mappings to improve kernel performance.

Here is the branch with the required changes.
https://github.com/atishp04/linux/tree/relocation_jal_issue

It works most of the time except when CONFIG_HARDENED_USERCOPY is enabled.
I see the following linker error in that case.

net/ipv4/ipconfig.o: in function `ic_setup_routes':
/home/atish/workspace/linux/net/ipv4/ipconfig.c:400:(.init.text+0x1c4): relocation truncated to fit: R_RISCV_JAL against symbol `ip_rt_ioctl' defined in .text section in net/ipv4/fib_frontend.o
make: *** [Makefile:1162: vmlinux] Error 1

This happens only when copy_from_user is called from function that is annotated with __init.
The relocation id seems RISCV_CALL from the objdump. I am not sure why linker converts it to JAL.
I further narrowed down the issue to a __builtin_constant_p

Here is the code path:
ip_rt_ioctl->rtentry_to_fib_config->copy_from_user->check_copy_size->check_object_size

```
#ifdef CONFIG_HARDENED_USERCOPY
112 extern void __check_object_size(const void *ptr, unsigned long n,
113                                         bool to_user);
114
115 static __always_inline void check_object_size(const void *ptr,
unsigned long n,
116                                               bool to_user)
117 {
118         if (!__builtin_constant_p(n)) ---------------------> Commenting this line avoids the linker issue
119         __check_object_size(ptr, n, to_user);
120 }
121 #else
122 static inline void check_object_size(const void *ptr, unsigned
long n,
123                                      bool to_user)
124 { }
125 #endif /* CONFIG_HARDENED_USERCOPY */

```

jim-wilson, 2020-10-15:
I tried a build on my fedora rawhide box, with gcc-10.2 and binutils-2.35.  I did not get a link error.  I can see from your config file that you have gcc-9.3, and binutils-2.33.  Checking ChangeLog entries, this could perhaps be 
    https://sourceware.org/bugzilla/show_bug.cgi?id=25181
fixed last November, which means it was fixed about a month and a half after the binutils-2.33 release.

atishp04, 2020-10-16:
@jim-wilson Did you try the kernel tree I shared ? I am getting linker error with configs generated with defconfig as well with gcc-10.2 & binutils-2.35. Earlier, I used to see the error only with the config shared.

init/main.o: in function `kernel_init':
main.c:(.ref.text+0xec): relocation truncated to fit: R_RISCV_JAL against `kernel_init_freeable'
mm/page_alloc.o: in function `.L0 ':
page_alloc.c:(.ref.text+0x46): relocation truncated to fit: R_RISCV_JAL against symbol `memblock_alloc_try_nid' defined in .init.text section in mm/memblock.o
drivers/base/devtmpfs.o: in function `.L0 ':
devtmpfs.c:(.ref.text+0x3c): relocation truncated to fit: R_RISCV_JAL against symbol `init_mount' defined in .init.text section in fs/init.o
drivers/firmware/efi/earlycon.o: in function `.L0 ':
earlycon.c:(.ref.text+0x24): relocation truncated to fit: R_RISCV_JAL against symbol `early_memremap_prot' defined in .init.text section in mm/early_ioremap.o
make: *** [Makefile:1162: vmlinux] Error 1

jim-wilson, 2020-10-16:
Yes, I tried your tree, but it didn't fail for me with gcc-10 and binutils-2.35.  I was able to reproduce last night using gcc-9.3 and binutils-2.33.1 though.  Adding the patch from 25181 didn't fix it, so I need to try to debug it.

atishp04, 2020-10-16:
Huh..I can't reproduce the issue with the config where CONFIG_HARDENED_USERCOPY is enabled. But it fails with the config generated using defconfig. I guess the fix works for some cases but fails for others.

Here are the steps to reproduce the issue with gcc-10 & binutils-2.35

1. Checkout the following branch
https://github.com/atishp04/linux/tree/relocation_jal_issue

2. Generate a config using defconfig
```
ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu- make defconfig
```
3. Compile Kernel

Here are the errors I see
```
init/main.o: in function `kernel_init':
main.c:(.ref.text+0xec): relocation truncated to fit: R_RISCV_JAL against `kernel_init_freeable'
mm/page_alloc.o: in function `.L0 ':
page_alloc.c:(.ref.text+0x46): relocation truncated to fit: R_RISCV_JAL against symbol `memblock_alloc_try_nid' defined in .init.text section in mm/memblock.o
drivers/base/devtmpfs.o: in function `.L0 ':
devtmpfs.c:(.ref.text+0x3c): relocation truncated to fit: R_RISCV_JAL against symbol `init_mount' defined in .init.text section in fs/init.o
drivers/firmware/efi/earlycon.o: in function `.L0 ':
earlycon.c:(.ref.text+0x24): relocation truncated to fit: R_RISCV_JAL against symbol `early_memremap_prot' defined in .init.text section in mm/early_ioremap.o
make: *** [Makefile:1162: vmlinux] Error 1
```

jim-wilson, 2020-10-20:
This is an alignment padding problem.  There is a call from .init.text to .text.  .init.text immediately follows .text.  The call is in range of a jal instruction when relaxed.  Further relaxation causes the .text target address to decrease.  An alignment directlve between .init.text and .text means that the .init.text starting address does not decrease during relaxation.  Normally, the difference between two code addresses can only decrease during relaxation.  But alignment directives means that offsets can increase, worst case, by as much as the specified alignment.  In this case, the offset ends up increasing just enough to make it too large for a JAL and we get a relocation overflow error at the end.

The linker tries to take alignment directives into account, by adding section alignment to the offset before deciding if it will fit in a relaxed instruction.  Normally alignments are small so this is OK.  In this case you have a large alignment.  But even worse is that the alignment is in between sections, and the RISC-V relaxation code isn't aware of it.  This causes the relocation error as we aren't adjust for alignments properly.  I don't know if we can handle alignment directives in between sections during relaxation time.  The relaxation code may not have access to such linker script info.  We do have access to section alignments though.

I can work around the problem by making the section alignments explicit in the linker script.  With this change
```
46c46
<  . = ALIGN((1 << (12))); .init.text : AT(ADDR(.init.text) - 0xffffffe000000000
) { _sinittext = .; *(.init.text .init.text.*) *(.text.startup) *(.meminit.text*
) _einittext = .; }
---
>  . = ALIGN((1 << (12))); .init.text : AT(ADDR(.init.text) - 0xffffffe000000000
) ALIGN(1 << 21) { _sinittext = .; *(.init.text .init.text.*) *(.text.startup) *
(.meminit.text*) _einittext = .; }
```
In the executable .init.text is now specified to have 1<<21 alignment.  it still has the same starting address as before.  The link succeeds with this change.  I get about a 6.6KB increase in the code size due to less relaxation, as the large section alignment will prevent a lot of cross section relaxation.  For calls within a section, we use that section's alignment in the calculation.  For calls across sections, e.g. from .init.text to .text, we use the max section alignment.  We really should use the max section alignment of the sections in between the source and target sections, but it isn't obvious how to compute that, so we just use the max of all section alignments which is safe.  This does mean that setting one section alignment to 1<<21 is going to affect all cross section relaxations though, and all relaxations inside that section.  Relaxations within .text will still happen as before though, and I only see the .text size increase by 32 bytes.  The .exit.text size increases by 100 bytes.  And .init.text increases by 6.5K.  Given that the kernel is about 8.9M a 6.5K increase should be OK.

atishp04, 2020-10-22:
Thanks @jim-wilson . That fixes the problem. I am not sure if that will be acceptable to upstream kernel though. I will propose this to the mailing list. Thanks a lot for the investigation. I have a question though. This may be a very stupid question as I don't have much experience in toolchain.

Why does the relaxation code doesn't have access to the alignment between sections while it have access to the section alignments ?

jim-wilson, 2020-10-22:
I know that the linker goes through the linker script commands in order as it operates, repeating this until the result stablizes, and I don't know if the relaxation code has access to any of that info.  But the relaxation code does have access to ELF stuff like section info, as that is all maintained in bfd not in the linker code.  So the linker scripts are handled by the code in the ld dir, and input and output sections contents and flags are handled by the code in the bfd dir.  The relaxation code is in the bfd dir also so it has access to all of the object file stuff, but maybe not the linker script.  I'm more of a gcc expert than a binutils expert, so I don't kinow if there is some obscure way to get the info we need to fix this.

atishp04, 2020-10-22:
Thanks @jim-wilson . @palmer-dabbelt any thoughts ?

lazyparser, 2021-03-04:
@atishp04 Hi, is this issue got solved?

TommyMurphyTM1234, 2022-09-27:
No update/activity in over a year so presuming that there's no outstanding issue at this stage.

closed 2022-09-27 by TommyMurphyTM1234

dinyy, 2024-04-08:
I want to known if this problem has been solved, I occur this proble when I want to cross-compile LLVM.
Log shows blow:

/home/sole/work/qemu_rvv/llvm-project-main/clang/lib/CodeGen/CGBuiltin.cpp:21038:(.text+0x13606a): relocation truncated to fit: R_RISCV_JAL against `.L13090'
collect2: error: ld returned 1 exit status
[442/486] Linking CXX static library lib/libclangSerialization.a
ninja: build stopped: subcommand failed.

dinyy, 2024-04-08:
> > I want to known if this problem has been solved, I occur this proble when I want to cross-compile LLVM. Log shows blow:
> > /home/sole/work/qemu_rvv/llvm-project-main/clang/lib/CodeGen/CGBuiltin.cpp:21038:(.text+0x13606a): relocation truncated to fit: R_RISCV_JAL against `.L13090' collect2: error: ld returned 1 exit status [442/486] Linking CXX static library lib/libclangSerialization.a ninja: build stopped: subcommand failed.
> 
> Please explain how exactly you are building LLVM - i.e. all commands/steps taken: `git clone`, `configure`, `make` etc. And please capture and attach the build log.

first, I get the newest LLVM by Download ZIP from github
second, my cmake command is that:
[image]
and my ToolchainExample.cmake shows blew:
[image]
And the error log when I building is that：
FAILED: lib/libclang-cpp.so.19.0git : && /home/sole/opt/riscv/gcc/bin/riscv64-unknown-linux-gnu-g++ -fPIC -march=rv64gcv -mcmodel=medany - ... ..... /home/sole/work/qemu_rvv/llvm-project-main/clang/lib/CodeGen/CGBuiltin.cpp:21038:(.text+0x13606a): relocation truncated to fit: R_RISCV_JAL against .L13090' collect2: error: ld returned 1 exit status [442/486] Linking CXX static library lib/libclangSerialization.a ninja: build stopped: subcommand failed.

dinyy, 2024-04-08:
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/sole/opt/riscv/gcc/bin/riscv64-unknown-linux-gnu-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/sole/opt/riscv/gcc/bin/riscv64-unknown-linux-gnu-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/sole/opt/riscv/gcc/bin/riscv64-unknown-linux-gnu-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- bolt project is disabled
-- clang project is enabled
-- clang-tools-extra project is disabled
-- compiler-rt project is disabled
-- cross-project-tests project is disabled
-- libc project is disabled
-- libclc project is disabled
-- lld project is enabled
-- lldb project is disabled
-- mlir project is disabled
-- openmp project is disabled
-- polly project is disabled
-- pstl project is disabled
-- flang project is disabled
-- Found Python3: /usr/bin/python3 (found suitable version "3.11.6", minimum required is "3.6") found components: Interpreter
-- Looking for dlfcn.h
-- Looking for dlfcn.h - found
-- Looking for errno.h
-- Looking for errno.h - found
-- Looking for fcntl.h
-- Looking for fcntl.h - found
-- Looking for link.h
-- Looking for link.h - found
-- Looking for malloc/malloc.h
-- Looking for malloc/malloc.h - not found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for signal.h
-- Looking for signal.h - found
-- Looking for sys/ioctl.h
-- Looking for sys/ioctl.h - found
-- Looking for sys/mman.h
-- Looking for sys/mman.h - found
-- Looking for sys/param.h
-- Looking for sys/param.h - found
-- Looking for sys/resource.h
-- Looking for sys/resource.h - found
-- Looking for sys/stat.h
-- Looking for sys/stat.h - found
-- Looking for sys/time.h
-- Looking for sys/time.h - found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for sysexits.h
-- Looking for sysexits.h - found
-- Looking for termios.h
-- Looking for termios.h - found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- Looking for valgrind/valgrind.h
-- Looking for valgrind/valgrind.h - found
-- Looking for fenv.h
-- Looking for fenv.h - found
-- Looking for FE_ALL_EXCEPT
-- Looking for FE_ALL_EXCEPT - found
-- Looking for FE_INEXACT
[... 249 lines cut ...]
-- Performing Test HAVE_CXX_FLAG_WSUGGEST_OVERRIDE
-- Performing Test HAVE_CXX_FLAG_WSUGGEST_OVERRIDE - Success
-- Performing Test HAVE_CXX_FLAG_PEDANTIC
-- Performing Test HAVE_CXX_FLAG_PEDANTIC - Success
-- Performing Test HAVE_CXX_FLAG_PEDANTIC_ERRORS
-- Performing Test HAVE_CXX_FLAG_PEDANTIC_ERRORS - Success
-- Performing Test HAVE_CXX_FLAG_WSHORTEN_64_TO_32
-- Performing Test HAVE_CXX_FLAG_WSHORTEN_64_TO_32 - Failed
-- Performing Test HAVE_CXX_FLAG_FSTRICT_ALIASING
-- Performing Test HAVE_CXX_FLAG_FSTRICT_ALIASING - Success
-- Performing Test HAVE_CXX_FLAG_WNO_DEPRECATED_DECLARATIONS
-- Performing Test HAVE_CXX_FLAG_WNO_DEPRECATED_DECLARATIONS - Success
-- Performing Test HAVE_CXX_FLAG_FNO_EXCEPTIONS
-- Performing Test HAVE_CXX_FLAG_FNO_EXCEPTIONS - Success
-- Performing Test HAVE_CXX_FLAG_WSTRICT_ALIASING
-- Performing Test HAVE_CXX_FLAG_WSTRICT_ALIASING - Success
-- Performing Test HAVE_CXX_FLAG_WD654
-- Performing Test HAVE_CXX_FLAG_WD654 - Failed
-- Performing Test HAVE_CXX_FLAG_WTHREAD_SAFETY
-- Performing Test HAVE_CXX_FLAG_WTHREAD_SAFETY - Failed
-- Performing Test HAVE_CXX_FLAG_COVERAGE
-- Performing Test HAVE_CXX_FLAG_COVERAGE - Success
-- Cross-compiling to test HAVE_GNU_POSIX_REGEX
-- Performing Test HAVE_GNU_POSIX_REGEX -- failed to compile
-- Configuring done (19.9s)

TommyMurphyTM1234, 2024-04-08:
What you've provided isn't sufficient for others to reproduce the issue. You need to ideally provide a small, self-contained, reproducible test case that will do this. 

You also didn't clarify all relevant details - how you built your RISC-V GCC toolchain, what specific LLVM tarball you downloaded to use, why you didn't build LLVM using these instructions, etc: 

* https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#llvm--clang

That's why I asked for details of all of the commands that you used from start to finish to build GCC/LLVM/Clang etc. 

And then a small, self-contained, reproducible test program that illustrates the error.

kito-cheng, 2024-04-08:
@dinyy could you file new issue for that? also plz give text version for the cmake command and ToolchainExample.cmake if possible?

dinyy, 2024-04-08:
> @dinyy could you file new issue for that? also plz give text version for the cmake command and ToolchainExample.cmake if possible?

OK,I'm trying to upload all this.

dinyy, 2024-04-08:
I file a new issue here：[https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1455](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1455) @TommyMurphyTM1234