riscv-collab/riscv-gnu-toolchain #739: Stack pointer modified when passing 2 word structure as function argument

kjetilos, 2020-10-14:
Here is a snippet of code that reproduces the issue

```C
#include <stdint.h>

struct block
{
   uint8_t *addr;
   uint32_t len;
};

static inline struct block block_convert(const void *array, uint32_t length)
{
    struct block blk = {(uint8_t *)array, length};
    return blk;
}

void write_block(struct block dest);

void get_block(uint8_t *dst, uint32_t size) {
    write_block(block_convert(dst, size));
}
```

This code generates instructions that allocate 16 bytes on the stack before the write_block function call. This allocation seems to be unnecessary.

```
$ riscv32-unknown-elf-gcc reference.c -march=rv32imac -mabi=ilp32 -msave-restore -o- -S -Os
        .file   "reference.c"
        .option nopic
        .attribute arch, "rv32i2p0_m2p0_a2p0_c2p0"
        .attribute unaligned_access, 0
        .attribute stack_align, 16
        .text
        .align  1
        .globl  get_block
        .type   get_block, @function
get_block:
        call    t0,__riscv_save_0
        addi    sp,sp,-16
        call    write_block
        addi    sp,sp,16
        tail    __riscv_restore_0
        .size   get_block, .-get_block
        .ident  "GCC: ('riscv32-embecosm-win64-gcc10.2.0-r3') 10.2.0"
```

Based on the code I would think that this could be optimized even further to a single call to write_block without even calling __riscv_save_0+__riscv_restore_0.

jim-wilson, 2020-10-15:
Stack space is allocated for temporaries, and then we are able to optimize away the temporaries.  But unallocating stack space is hard as that leads to aliasing issues, it is very hard to prove that there are no longer any stack frame references.  In a simple case like this it is possible.  But hard in general.  And optimizing the simple case requires some special purpose code that won't be very elegant and probably won't be very useful, as this problem likely doesn't occur very often.

Not a RISC-V specific problem, and not related to -msave-restore.

kjetilos, 2020-10-16:
I see something similar for arm, so it looks like a non risc-v specific issue. Here is the generated code for a Cortex-M4

```
get_block:
        sub     sp, sp, #8
        strd    r0, r1, [sp]
        add     r3, sp, #8
        ldmdb   r3, {r0, r1}
        add     sp, sp, #8
        b       write_block
```

closed 2021-03-04 by lazyparser