riscv-collab/riscv-gnu-toolchain #740: double copies of structure on stack when calling function with struct argument

kjetilos, 2020-10-14:
This can be reproduced with the following code. Here we create a structure with the size of 24 bytes and pass it to a function. The generated code will create two copies of this structure in the stack before calling the write_block function.

```C
#include <stdint.h>

struct block
{
   uint8_t *addr;
   uint32_t len;
   uint32_t flags;
};

void write_block(struct block dest);

void get_block(uint8_t *dst, uint32_t size) {
    struct block blk = {dst, size, 0x0};
    write_block(blk);
}
```

And here is the generated code when building with -Os

```
$ riscv32-unknown-elf-gcc reference2.c -march=rv32imac -mabi=ilp32 -msave-restore -o- -S -Os
        .file   "reference2.c"
        .option nopic
        .attribute arch, "rv32i2p0_m2p0_a2p0_c2p0"
        .attribute unaligned_access, 0
        .attribute stack_align, 16
        .text
        .align  1
        .globl  get_block
        .type   get_block, @function
get_block:
        call    t0,__riscv_save_0
        addi    sp,sp,-32
        sw      a0,20(sp)
        sw      a1,24(sp)
        li      a2,12
        addi    a1,sp,20
        mv      a0,sp
        sw      zero,28(sp)
        call    memcpy
        mv      a0,sp
        call    write_block
        addi    sp,sp,32
        tail    __riscv_restore_0
        .size   get_block, .-get_block
        .ident  "GCC: ('riscv32-embecosm-win64-gcc10.2.0-r3') 10.2.0"
```

jim-wilson, 2020-10-15:
The local struct blk won't fit in registers so needs to be allocated to the stack.  Then it is passed by value to get_block, which requires a copy, because we can't let it modify the local.  The RISC-V calling convention says the arg must be passed on the stack, so we need to create a second copy on the stack.

Since the local dies in the call, in theory we can just have one copy on the stack, but this is complicated by the fact that calling conventions are target dependent and not exposed at the high level gimple IL, and hence can't be optimized there.  And at the low level RTL where we can see the problem, we then have to deal with alias issues which makes this very hard in general.

I see similar issues just looking at arm and x86_64 code, except they are both passing the arg in registers, and hence are able to avoid the second copy on the stack.  Probably other targets like mips will show the same second copy on the stack.

Looks hard to fix.

kjetilos, 2020-10-16:
Thanks for the explanation, that makes sense. The best solution in my case might be to refactor the source code to not use this pattern.

closed 2021-03-04 by lazyparser