riscv-collab/riscv-gnu-toolchain #742: Linker error

marethyu, 2020-10-18:
I was trying to install riscv toolchain from the source. After executing the makefile, I got the following linker error
```
...
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/10.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /mnt/c/users/jimmy/downloads/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os: in function `__GI___assert_perror_fail':
/mnt/c/users/jimmy/downloads/riscv-gnu-toolchain/riscv-glibc/elf/dl-minimal.c:224: undefined reference to `rtld_errno'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/10.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /mnt/c/users/jimmy/downloads/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os: in function `__strsep':
/mnt/c/users/jimmy/downloads/riscv-gnu-toolchain/riscv-glibc/elf/dl-minimal.c:272: undefined reference to `rtld_errno'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/10.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /mnt/c/users/jimmy/downloads/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os:/mnt/c/users/jimmy/downloads/riscv-gnu-toolchain/riscv-glibc/elf/dl-minimal.c:279: more undefined references to `rtld_errno' follow
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/10.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /mnt/c/users/jimmy/downloads/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/elf/ld.so.new: hidden symbol `__libc_check_standard_fds' isn't defined
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/10.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:496: /mnt/c/users/jimmy/downloads/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/elf/ld.so] Error 1
make[3]: Leaving directory '/mnt/c/users/jimmy/downloads/riscv-gnu-toolchain/riscv-glibc/elf'
make[2]: *** [Makefile:258: elf/subdir_lib] Error 2
make[2]: Leaving directory '/mnt/c/users/jimmy/downloads/riscv-gnu-toolchain/riscv-glibc'
make[1]: *** [Makefile:9: all] Error 2
make[1]: Leaving directory '/mnt/c/users/jimmy/downloads/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d'
make: *** [Makefile:247: stamps/build-glibc-linux-rv64imafdc-lp64d] Error 2
```
I have no idea what happened. I didn't expect the error because I followed the instructions carefully. Can anyone tell me why I got this error and how to resolve it?

I am using WSL2.

lazyparser, 2021-03-04:
@marethyu Hi, have you solved it?

marethyu, 2021-03-04:
@lazyparser I gave up, unfortunately.

closed 2021-03-05 by lazyparser

shenshouer, 2022-02-24:
I met the same error :

```
/mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/riscv-glibc/elf/dl-close.c:824: undefined reference to `__lll_lock_wait_private'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os: in function `glibc_hwcaps_priorities_init':
/mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/riscv-glibc/elf/dl-cache.c:97: undefined reference to `__libc_check_standard_fds'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os: in function `_dl_catch_exception':
/mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/riscv-glibc/elf/dl-error-skeleton.c:219: undefined reference to `rtld_errno'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os: in function `_dl_catch_error':
/mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/riscv-glibc/elf/dl-error-skeleton.c:225: undefined reference to `rtld_errno'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os: in function `_dl_resolve_conflicts':
/mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/riscv-glibc/elf/dl-conflict.c:55: undefined reference to `rtld_errno'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os: in function `elf_machine_rela':
/mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/riscv-glibc/elf/../sysdeps/riscv/dl-machine.h:187: undefined reference to `rtld_errno'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os: in function `copy_hwcaps':
/mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/riscv-glibc/elf/dl-hwcaps.c:90: undefined reference to `rtld_errno'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os:/mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/riscv-glibc/elf/dl-hwcaps.c:184: more undefined references to `rtld_errno' follow
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/elf/ld.so.new: hidden symbol `__libc_check_standard_fds' isn't defined
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:582: /mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/elf/ld.so] Error 1
make[3]: Leaving directory '/mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/riscv-glibc/elf'
make[2]: *** [Makefile:479: elf/subdir_lib] Error 2
make[2]: Leaving directory '/mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/riscv-glibc'
make[1]: *** [Makefile:9: all] Error 2
make[1]: Leaving directory '/mnt/d/workspaces/os/risc-v-toolchain/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d'
make: *** [Makefile:341: stamps/build-glibc-linux-rv64imafdc-lp64d] Error 2
```

jim-wilson, 2022-03-07:
This may be a problem with case sensitive file names.  GLIBC requires a case sensitive file system in order to build correctly.  All unix systems have case sensitive file names.  Windows by default does not have case sensitive files.  An attempt to build on a non-case sensitive file system will fail with confusing errors.  I don't know what the default for WSL2 is.  But maybe your WSL2 is inside a non-case sensitive file system, in which case a glibc build will not work.  The linux kernel also requires a case sensitive file system, as otherwise some linux kernel header files will have conflicting names.

GramThanos, 2022-11-18:
@jim-wilson thanks for pointing this out. I want that your comment solved my problem.
In WSL2 one may check if the file system on a particular folder is case sensitive using `getfattr -n system.wsl_case_sensitive ./`.
For me, the my WSL2's user home if case sensitive (e.g. `~/myproject`), while all the other external folder are not (e.g. `/mnt/d/myproject/`).