riscv-collab/riscv-gnu-toolchain #743: unrecognized opcode error

shreyajarsania, 2020-10-19:
Hello,
I have created a simple instruction in riscv-llvm. I have used https://github.com/andestech/riscv-llvm-toolchain toolchain . A code with my instruction compiles when done with riscv32-unknown-elf-clang but give **unrecognized opcode error** when done with riscv32-unknown-elf-gcc. Is there a method where i can add my instruction in riscv-unknown-elf-gcc? It gives same error when tried with riscv32-unknown-elf-as.

Thank you
Shreya Jarsania

kito-cheng, 2020-10-20:
I guess you are adding instruction to LLVM only, if you want `gcc` and `as`, you need to add the instruction into `binutils`.

shreyajarsania, 2020-10-20:
Hello,
Thank for your response. Yes I was adding it in llvm only. I will surely do this.

Thank you.
Shreya Jarsania

shreyajarsania, 2020-11-02:
Hii,
I want to link the llvm-riscv instruction with c language code. But while adding an instruction in riscv-binutils-gdb(riscv-gnu-toolchain/riscv-binutils-gdb/include/opcode/riscv-opc.h), it needs some MATCH and MASK ids. How do i get it for a riscv-llvm instructioni or how do i link it there?

jim-wilson, 2020-11-02:
MASK has a 1 bit for every non-operand bit in the encoding, and MATCH has a 1 bit for every non-operand bit that must be 1.  So on disassembly we do (insn & MASK) == MATCH to see if the instruction matches this line.  On assembly, we or in MATCH, and then or in the operand values to get the instruction encoding.

shreyajarsania, 2020-11-03:
Thank you for explanation. I will try this form.

lazyparser, 2021-03-04:
@shreyajarsania Hi, do we have any updates on this issue?

TommyMurphyTM1234, 2022-09-27:
No update/activity in over a year and no issue with the RISC-V GCC toolchain itself identified.

closed 2022-09-27 by TommyMurphyTM1234