riscv-collab/riscv-gnu-toolchain #744: build process error:

AnandS09, 2020-10-21:


AnandS09, 2020-10-21:
While trying to build on ubuntu 18.04 encountered the following error. 
g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fnoo-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qa-rtrlengual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-vaamigcc/griadic-macros -Wno-overlength-strings  -Wno-unused -DHAVE_CONFIG_H -I. -I. -I../../.scro.././asynchronous-unwind-/riscv-gcc/gcc -I../.././riscv-gcc/gcc/. -I../.././riscv-gcc/gcc/../include -I../../sg -ber -ributeverled-iua-pantables -W -Wall -Wnoverlen./riscv-gcc/gcc/../libcpp/include  -I../.././riscv-gcc/gcc/../libdecnumber -I../../.vc/lMF ./-Wno-uHAVEG_HI./.rico-llongvariadcros -I../ree)’/riscv-gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././riscv-gcc/gcc/../libbcgecacktrace   -o gimple-match.o -MT gimple-match.o -MMD -MP -MF ./.deps/gimple-match.TPe.ou//lbe././v-gbdeciscv-gcnumbo gimple-match.c
gimple-match.c: In function ‘bool gimple_simplify_VEC_COND_EXPR(gimple_match_op*, gimplemple**, tree_node* (*)(tree), code_helper, tree, tree, tree, tree)’:
gimple-match.c:119283:12: internal compiler error: in extract_ops_from_tree, at gimple-ele-expr.c:556
       _q32 = do_valueize (valueize, _q32);
       ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Please submit a full bug report,
with preprocessed source if appropriate.

I have followed the steps mentioned in the readme for riscv-gnu-toolchain installation. 

Any help is highly appreciated.

kito-cheng, 2020-10-21:
It seems your g++ something wrong, could you provide more info? e.g. version for `g++`, which can be observed by `g++ -v`.

And according your description, does it mean it's OK to build if you don't give `--enable-multilib` option?

AnandS09, 2020-10-21:
Hi @kito-cheng, 

Thank you very much for your reply. 
This is the version of the gcc which threw the error `gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)`
I think this is the latest version. 
`g++ is already the newest version (4:7.4.0-1ubuntu2.3)`

>And according your description, does it mean it's OK to build if you don't give --enable-multilib option?
Sorry for the confusion, I just tried without this option and encountered the same error.

jim-wilson, 2020-10-21:
I can do a toolchain build on my Ubuntu 18.04 laptop with no problem.  I have the same gcc version.  Looks like you got the gcc version from gcc -v and the g++ version from apt upgrade which is confusing.  To get the right version number from apt you need to check the g++-7 package not the g++ package.  Or just use g++ -v which is simpler.

You should verify that your source tree is up to date, e.g. try "git status" and see if it complains about anything.

There are some less common things that can break a build such as defining unusual and unexpected environment variables that change compiler behavior.

gimple-match.c is one of the largest source files, but it needs less than 1G to compile.  If you only have 0.5G mem+swap that could be a problem, but seems unlikely.

AnandS09, 2020-10-21:
Hi @jim-wilson ,

Thank you for your inputs. Let me try to retrace the steps. 
I am appending my console outputs so that we can rule out any obvious errors. 

**g++ version**
>anand@airchitect:~/local/riscv/riscv-gnu-toolchain$ g++ -v
`Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.5.0-3ubuntu1~18.04' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04) `

**Git status**
>anand@airchitect:~/local/riscv/riscv-gnu-toolchain$ git status .
`On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean`

I will try start from a fresh directory and post what happens, just in case there are any problems with the paths.

lazyparser, 2021-03-04:
@AnandS09 Hi, have you solved this issue?

TommyMurphyTM1234, 2022-09-27:
No update/activity for over a year.

closed 2022-09-27 by TommyMurphyTM1234