riscv-collab/riscv-gnu-toolchain #745: Getting Issues on make on Mac ( Catalina)

johnnavaj, 2020-10-21:
[image]

jim-wilson, 2020-10-21:
sim-core.c includes stdlib.h, and stdlib.h declares free.  So this is odd.  I don't have access to a mac so I can't try to reproduce.

I do know that building on a mac requires a few extra steps, as you have to use GNU tools instead of default Mac tools, e.g. gsed instead of sed, gtar instead of tar, etc.  SiFive builds our toolchains on macos, and we have a makefile to handle this.  You might want to try looking at github.com/sifive/freedom-tools to see how it makes the macos build work.  I don't think that this is related to the error that you got though.

johnnavaj, 2020-10-22:
Hi jim , 
Looks like free has been moved from stdlib.h to _malloc.h hence this throws this . So looks like the CTL for Mac OS  has been updated from the last version I was using. ( Which had free in stdlib.h).

So someone will have to update this.

johnnavaj, 2020-10-22:
Using prebuilts for now, I mainly wanted to build a soft float capable toolchain so ran into this. Seems that this toolchain was entirely tested and verified for OS X ,so can't really say between the resource difference between Catalina and OS X either in terms of the GNU tools etc provided by the CTL.

jim-wilson, 2020-10-22:
The ISO C standard says that free is declared in stdlib.h.  There is something wrong with your system if it isn't there.  Though maybe something needs to be enabled or defined for stdlib.h to include _malloc.h?

tyfkda, 2020-11-11:
@johnnavaj 
I have encountered same problem.
I thinks it is caused because `riscv-gdb` submodule is old.
When I use newer one, I have succeeded to `make` in this repository.

@jim-wilson chould you please take a look #769 ? Thanks.

johnnavaj, 2020-11-13:
@tyfkda Yes I can confirm this , looks like the gdb needs a bump in its submodule.

tyfkda, 2020-11-20:
@johnnavaj I think this issue is solved #776

johnnavaj, 2020-11-21:
@tyfkda thanks for this

closed 2020-11-21 by jim-wilson