riscv-collab/riscv-gnu-toolchain #746: Using -fopenmp argument causes error: unrecognized command-line option '-pthread' (How do you compile with OpenMP?)

JL102, 2020-10-22:
I'm attempting to compile some OpenMP programs in the Rodinia benchmark suite (http://rodinia.cs.virginia.edu/doku.php) for RISC-V. 
I'm starting with a relatively small benchmark with a single file (srad_v2, which can be seen here: https://github.com/JL102/rowan-riscv-rodinia-llvm/blob/main/src/openmp/srad/srad_v2/srad.cpp).

When I attempt to compile it with the `-fopenmp` argument, I get this error:
```
$ riscv64-unknown-elf-g++ -fopenmp -o srad_riscv srad.cpp
riscv64-unknown-elf-g++: error: unrecognized command-line option '-pthread'
compilation terminated.
```
and when I attempt to compile it with the `-openmp` argument, I get this error:
```
$ riscv64-unknown-elf-g++ -openmp -o srad_riscv srad.cpp
srad.cpp:13:10: fatal error: omp.h: No such file or directory
   13 | #include <omp.h>
      |          ^~~~~~~
compilation terminated.
```

I'm not surprised that compiling for RISC-V with OpenMP is not as simple as adding a couple command-line arguments. But I have no idea how to, since there doesn't seem to be any documentation yet on OpenMP support for the riscv-gnu-toolchain.

aswaterman, 2020-10-22:
OpenMP requires pthreads, and pthreads is not provided for the ELF backend.  If you use a Linux compiler (riscv64-unknown-linux-gnu-g++) then what you're trying to do should work.

JL102, 2020-10-22:
Okay, thanks. Could you point me to some info on the difference between ELF and Linux compilation? That is, I don't really understand what the difference is in the end product or how they run. Do riscv64-unknown-linux-gnu-g++ executables still need to be executed the exact same way with Spike and PK, as shown in the example on riscv-isa-sim's readme?

aswaterman, 2020-10-22:
In short, the ELF compiler is for bare-metal programs, and the Linux compiler is for applications targeting the Linux ABI.

I think using the ELF compiler for targeting PK is the right thing to do.  A subset of executables generated by the Linux compiler might also run on PK.  But note that PK doesn't support the pthreads-related system calls, so even if you generate a working executable, you won't be able to run it there.  You'll need to actually boot Linux, which is doable but is outside the scope of what we can support here.

JL102, 2020-10-22:
Ahhh, I see. That's the answer I was fearing - since I don't know if we want to make a processor that boots linux.

By any chance, is OpenCL any different in this context? I haven't fiddled with it at all, but most of the same benchmarks are also written using OpenCL and I often see the two being compared.

aswaterman, 2020-10-22:
I know POCL has a pthread backend (which obviously has the same limitations).  It might be practical to port POCL to bare-metal RISC-V; I don't know.

Whatever the case is, PK won't be of help, because only supports single-threaded processes.  So you'd probably need to actually run bare metal, rather than running on top of PK.  If you want an example of a bare-metal C program that runs on Spike, well, PK itself is a good example.

I suspect whatever you do will take a fair amount of work.  It's obviously a bit of a challenge to build a processor that can run Linux, but the reward is you get its universe of software.

JL102, 2020-10-22:
Thank you for the info. I'll close this thread since you did answer my question, but can I still come to you here for questions?

I'll try doing more research on those libraries. Re: porting POCL to bare-metal RISC-V, would that be possible if it requires pthread?

closed 2020-10-22 by JL102

aswaterman, 2020-10-22:
Questions that pertain to the toolchain, for sure.  You might want to use the sw-dev mailing list (https://groups.google.com/u/2/a/groups.riscv.org/g/sw-dev) for other stuff.