riscv-collab/riscv-gnu-toolchain #747: Add prebuilt releases for different platforms

Banyc, 2020-10-22:
It is too expensive to build the toolchain in some machines. It includes massive time consumption

- on cloning the whole repository,
- on the building process, 
- and on possible trouble-shootings.

If the prebuilt releases are available, it has advantages of:

- reducing the costs listed above,
- reducing the total time consumption spread to every end-users.
- and lowering the threshold for RISC-V development, attracting many new-comers to join the RISC-V community.

SiFive has provided a toolchain for the x86_64 platform, but the other platforms are still not available.

So, the suggestion might be adding prebuilt releases for different platforms.

Banyc, 2020-10-22:
...or consider adding apt download source XD

kito-cheng, 2020-10-22:
@kevinpark1217 has contributed the github action for pre-built things, https://github.com/riscv/riscv-gnu-toolchain/pull/710, but I guess need some one or maybe some github action script to publish into [release page](https://github.com/riscv/riscv-gnu-toolchain/releases).

> ...or consider adding apt download source XD
Sounds good to me, but it need some volunteer to do that.

jim-wilson, 2020-10-22:
RISC-V International doesn't have any engineering staff, so no easy way to provide precompiled toolchains.

Some RISC-V International members provide precompiled copies of their toolchains.  SiFive for instance has riscv64-elf toolchains on its website to support SiFive products.  These are built for Ubuntu, CentOS, MS Windows, and Mac.  CodeSourcery provides precompiled riscv64-elf toolchains on their web site.

Debian does have a riscv64-elf toolchain package.  Bits of it appeared in ubuntu 20.04 but it appears to be incomplete last I checked.  You would need a recent version of Debian to get it.

Most linux distros are now providing riscv64-linux cross compilers.  Ubuntu 18.04 has gcc v7 and v8 based cross compilers for riscv64-linux.  Ubuntu 20;.04 has gcc v8, v9, and v10 riscv64-linux cross compiles.

kevinpark1217, 2021-01-12:
> I guess need some one or maybe some github action script to publish into release page.

I'm currently working on a GitHub action that will do this automatically.
Although I would like some help on how to name prebuilt packages. And which flags in `./configure` makes it so.
For example, what does it exactly mean by `unknown` in the name below.

For examples,
`riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-apple-darwin.tar.gz`  
`riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14.tar.gz`

Banyc, 2021-01-25:
The nightly builds are amazing! Thanks for the efforts! I will close this issue.

closed 2021-01-25 by Banyc