riscv-collab/riscv-gnu-toolchain #749: rvv-intrinsic branch can't build

edre, 2020-10-26:
From the instructions in the announcement email: https://lists.riscv.org/g/tech-vector-ext/topic/76328927#364

```
$ git clone https://github.com/riscv/riscv-gnu-toolchain -b rvv-intrinsic
$ cd riscv-gnu-toolchain; mkdir build; cd build
$ ../configure --with-arch=rv64gcv_zfh --prefix=$RISCV
$ make newlib build-qemu
/bin/sh: 1: riscv-gnu-toolchain/build/../riscv-binutils/configure: not found
make: *** [Makefile:398: stamps/build-binutils-newlib] Error 127
```
Guess we need to add submodules.
```
$ git submodule update --init
...
Submodule 'qemu' (git@github.com:sifive/qemu.git) registered for path '../qemu'
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.
```
So that should be updated to a public repo. Ignoring qemu for now.
```
$ git submodule update --init ../riscv-newlib ../riscv-gcc
$ make newlib
...
g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../../riscv-gcc/gcc -I../../../riscv-gcc/gcc/. -I../../../riscv-gcc/gcc/../include -I../../../riscv-gcc/gcc/../libcpp/include  -I../../../riscv-gcc/gcc/../libdecnumber -I../../../riscv-gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../../../riscv-gcc/gcc/../libbacktrace   -o insn-emit.o -MT insn-emit.o -MMD -MP -MF ./.deps/insn-emit.TPo insn-emit.c
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [Makefile:1117: insn-emit.o] Error 1
make[2]: Leaving directory 'riscv-gnu-toolchain/build/build-gcc-newlib-stage1/gcc'
make[1]: *** [Makefile:4388: all-gcc] Error 2
make[1]: Leaving directory 'riscv-gnu-toolchain/build/build-gcc-newlib-stage1'
make: *** [Makefile:440: stamps/build-gcc-newlib-stage1] Error 2
```

I successfully built this a few months ago on the same machine (with 4GB RAM).

kito-cheng, 2020-10-27:
The monster require about ~7G memory to compile on our ubuntu 18.04 server with gcc 7.5, so I guess you might need add more swap memory to build that...I think this issue should resolve before upstream, but we don't have enough bandwidth to do that.

edre, 2020-10-29:
I successfully got it to build on a machine with more RAM. The qemu submodule path should still be updated.

kito-cheng, 2020-10-29:
I just push a repo path fix for qemu https://github.com/riscv/riscv-gnu-toolchain/commit/3fa82f8882b1af251382bf12ba62db0d3ec1ef20, could you try it?

edre, 2020-10-29:
Git makes it painful to switch to the new submodule path for an existing repo, but I successfully picked up the change on a fresh clone.

closed 2020-10-29 by edre