riscv-collab/riscv-gnu-toolchain #751: linker relax can cause address issue resulting in seg fault

rgdaddio, 2020-10-27:
In this simple 'hello world' when the second symbol (test) is loaded it will have an incorrect address and seems to incorrectly remove an instruction(optimize out?). This may be part of other riscv linker relax behaviors. 

	.data

myStr:
	.string "hello riscv new world!!\n" 	#;basic string data
	len = . - myStr                 	       #;get the length

test:
	.string "hello riscv new world2!!\n" 	#;basic string data
	tlen = . - test                 	                #;get the length
//snip
_start:
 //loads correct address first symbol
	lui a1, %hi(myStr)		               #high part of message
	addi a1, a1, %lo(myStr)		       #low part of messate
//loads wrong address for second symbol(test)
	**lui a1, %hi(test)		            #high part of message
	addi a1, a1, %lo(test)		    #low part of message**
	li a2, tlen			                    #len    

objdump
   100b0:       00100513                li      a0,1
   100b4:       000115b7                lui     a1,0x11
   100b8:       0fc58593                addi    a1,a1,252 # 110fc <__DATA_BEGIN__>
   100bc:       01900613                li      a2,25
   100c0:       00000693                li      a3,0
   100c4:       04000893                li      a7,64
   100c8:       00000073                ecall
   **100cc:       00100513                li      a0,1
   00d0:       81918593                addi    a1,gp,-2023 # 11115 <test>**

If the code is modified to use the norelax option on the second symbol the correct address is loaded

	.data

myStr:
	.string "hello riscv new world!!\n" 	#;basic string data
	len = . - myStr                 	        #;get the length

	**.option norelax**
test:
	.string "hello riscv new world2!!\n" 	#;basic string data
	tlen = . - test                 	                #;get the length

objdump
   100b0:       00100513                li      a0,1
   100b4:       000115b7                lui     a1,0x11
   100b8:       10058593                addi    a1,a1,256 # 11100 <__DATA_BEGIN__>
   100bc:       01900613                li      a2,25
   100c0:       00000693                li      a3,0
   100c4:       04000893                li      a7,64
   100c8:       00000073                ecall
   100cc:       00100513                li      a0,1
   **100d0:       000115b7                lui     a1,0x11
   100d4:       11958593                addi    a1,a1,281 # 11119 <test>**

GNU ld (GNU Binutils) 2.35
riscv64-unknown-linux-gnu-as hello_world_riscv_test.s -o hello_world_riscv_test.o
riscv64-unknown-linux-gnu-ld hello_world_riscv_test.o -o hello_world_riscv_test

This was tested with Spike and pk. Code with this same format in x86, arm, openrisc assembly all work with initial code no option needed. This may be the expected linker relax/optimize behavior with riscv and might not occur in more general coding environments.

jim-wilson, 2020-10-27:
>00d0: 81918593 addi a1,gp,-2023 # 11115

Why do you think that this is incorrect?  It looks OK.

Did you load the global pointer value into the gp register?  This needs to be done at program entry.  Newlib _start has
_start:
  # Initialize global pointer
.option push
.option norelax
1:auipc gp, %pcrel_hi(__global_pointer$)
  addi  gp, gp, %pcrel_lo(1b)
.option pop

jim-wilson, 2020-10-27:
Part 3 of Palmer Dabbelt's blog on the RISC-V toolchain talks about linker relaxation and the GP register.
https://www.sifive.com/blog/all-aboard-part-3-linker-relaxation-in-riscv-toolchain
You might be interested in reading other parts of this blog also.

rgdaddio, 2020-10-27:
Thanks.

I didn't use the gp at all in the code so didn't initialize. The linker inserted the use of the gp and apparently optimized out the lui (as part of relax?). 

The problem I encountered was that this caused a seg fault when the code ran in spike.  Using the initialization code from newlib uses norelax option which is what I used as a workaround (shown above as **.option norelax**). 
  
Saw Palmer Dabbelt's blog that is what made me think this might be unique to riscv.

jim-wilson, 2020-10-27:
MIPS also has a global pointer register that must be initialized in _start.  The .norelax in the newlib _start is just to avoid relaxation of the code to load the initial gp value.  Relaxation is turned back on after we init the gp register.  Other targets use linker relaxation.  The RISC-V uses relaxation to produce gp relative references, which may be unique to RISC-V.

rgdaddio, 2020-10-27:
Ok yes validated that by actually making sure to turn relax back on after initialization things worked fine. Thanks!!

rgdaddio, 2020-10-28:
Proper initialization of gp required.

closed 2020-10-28 by rgdaddio