riscv-collab/riscv-gnu-toolchain #753: Getting error '--64' command line option not supported

rajat-agnisys, 2020-10-28:
Facing this issue while installing Riscv tool chain:

g++ -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  -DGENERATOR_FILE -fno-PIE -I. -Ibuild -I../.././riscv-gcc/gcc -I../.././riscv-gcc/gcc/build -I../.././riscv-gcc/gcc/../include  -I../.././riscv-gcc/gcc/../libcpp/include  \
	-o build/genmodes.o ../.././riscv-gcc/gcc/genmodes.c
/opt/riscv64/riscv64-unknown-elf/bin/as: unrecognized option '--64'
make[2]: *** [Makefile:2724: build/genmodes.o] Error 1
make[2]: Leaving directory '/home/rajat/riscv/riscv-gnu-toolchain/build-gcc-newlib-stage1/gcc'
make[1]: *** [Makefile:4388: all-gcc] Error 2
make[1]: Leaving directory '/home/rajat/riscv/riscv-gnu-toolchain/build-gcc-newlib-stage1'

Tried setting "PATH=/opt/riscv/bin" but did not get out of this issue 
I am using centos-release-8.2-2.2004.0.2.el8.x86_64 

Thanks

kito-cheng, 2020-10-28:
I guess it's some thing wrong in your PATH, could you create a new terminal and remove whole build folder and try again?

jim-wilson, 2020-10-28:
Looks like you put $prefix/$target/bin on your path which is a bad idea.  This contains a riscv as program, and now the native x86 compiler is trying to use it which won't work.  The programs in $prefix/$target/bin are there for the cross gcc to use, not for users.  Users should be putting $prefix/bin on their path instead, and using riscv64-uknown-elf-as as the assembler, so avoid confusion with the native assembler.

rajat-agnisys, 2020-10-29:
Thanks 

but now I am getting this issue..

g++ -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  -DGENERATOR_FILE -fno-PIE -I. -Ibuild -I../.././riscv-gcc/gcc -I../.././riscv-gcc/gcc/build -I../.././riscv-gcc/gcc/../include  -I../.././riscv-gcc/gcc/../libcpp/include  \
	-o build/genmodes.o ../.././riscv-gcc/gcc/genmodes.c
as: line 106: exec: -I: invalid option
exec: usage: exec [-cl] [-a name] [command [arguments ...]] [redirection ...]
make[2]: *** [Makefile:2724: build/genmodes.o] Error 1
make[2]: Leaving directory '/home/rajat/riscv/riscv-gnu-toolchain/build-gcc-newlib-stage1/gcc'
make[1]: *** [Makefile:4388: all-gcc] Error 2
make[1]: Leaving directory '/home/rajat/riscv/riscv-gnu-toolchain/build-gcc-newlib-stage1'
make: *** [Makefile:440: stamps/build-gcc-newlib-stage1] Error 2

jim-wilson, 2020-10-29:
Now it looks like the problem is that you have dot "." on your path.  The cross gcc creates a link or script to the cross as in the build dir for its use.  But if you have dot on your path then the native gcc will accidentally try to use it and will fail.    Though I don't understand why you got an error from exec, hopefully that isn't important.

Note that having a path that starts or ends with a colon, or has two colons in a row, is equivalent to having dot on your path.

Also note having dot on your path is a protential security risk.  If an attacker can trick you into a cd into a writable dir that contains a hacked ls program, then running ls in this dir will give the attacker control of your account.  So generally you should not put dot on your path, and if you want to run a program in the current directory then you type ./a.out for instance.  If this is too inconvenient, then put dot at the end of your path, so common programs like as can't be overridden by an attacker.

rajat-agnisys, 2020-10-31:
Hello again...

Now I am facing this issue..

Fatal error: invalid -march= option: `rv32i'
make[9]: *** [Makefile:393: lib_a-dummy.o] Error 1

Thanks..

rajat-agnisys, 2020-10-31:
make[9]: Entering directory '/riscv/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/rv32i/ilp32/newlib/libc/argz'
riscv64-unknown-elf-gcc -B/riscv/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/rv32i/ilp32/newlib/ -isystem /riscv/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/rv32i/ilp32/newlib/targ-include -isystem /riscv/riscv-gnu-toolchain/riscv-newlib/newlib/libc/include -B/riscv/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/rv32i/ilp32/libgloss/riscv64 -L/riscv/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/rv32i/ilp32/libgloss/libnosys -L/riscv/riscv-gnu-toolchain/riscv-newlib/libgloss/riscv64  -march=rv32i -mabi=ilp32 -DPACKAGE_NAME=\"newlib\" -DPACKAGE_TARNAME=\"newlib\" -DPACKAGE_VERSION=\"3.2.0\" -DPACKAGE_STRING=\"newlib\ 3.2.0\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -I. -I/riscv/riscv-gnu-toolchain/riscv-newlib/newlib/libc/argz -fno-builtin -DHAVE_NANOSLEEP      -O2 -D_POSIX_MODE   -mcmodel=medlow  -march=rv32i -mabi=ilp32 -c -o lib_a-dummy.o `test -f 'dummy.c' || echo '/riscv/riscv-gnu-toolchain/riscv-newlib/newlib/libc/argz/'`dummy.c
Assembler messages:

In this command.

jim-wilson, 2020-10-31:
I would suggest you cd into the dir where the command was run, and then rerun it with a -v (verbose) option.  This will show what assembler the compiler is calling, and what options are being passed to it.

PaoloFR, 2020-11-03:
Having similar problems:

First had the "--64" problem and fixed according to the suggestions above.  Now I am having the following error:

checking for suffix of object files... configure: error: in `/TAREA/opt/install/riscv-gnu-toolchain/build-gcc-linux-stage1/riscv64-unknown-linux-gnu/libgcc':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
make[1]: *** [Makefile:11616: configure-target-libgcc] Error 1
make[1]: Leaving directory '/TAREA/opt/install/riscv-gnu-toolchain/build-gcc-linux-stage1'
make: *** [Makefile:276: stamps/build-gcc-linux-stage1] Error 2

My $PATH is: /usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin

The product directory that includes configure and Makefile is: /TAREA/opt/install/riscv-gnu-toolchain

PaoloFR, 2020-11-03:
Found out the problem.  The directory containing the Makefile had the "install" subdirectory in its full path.
I changed that and everything worked.

closed 2021-03-04 by lazyparser