riscv-collab/riscv-gnu-toolchain #754: When I call the C language library function, it shows undefined reference to

hqw-lantian, 2020-10-29:
My main function needs to call a random function (rand), I added include<stdlib.h> to the header file, but I call riscv64-unknown-elf-gcc to compile the main function, but the error is undefined reference to
The specific mian function is as follows:
#include "reg.h"
#include "pub_def.h"
#include "nvic.h"
#include "encoding.h"
#include "gpio.h"
#include "pub.h"

#include <stdio.h>
#include <stdlib.h>
#include <time.h>

int main(void)
{
	int i = 0;
 	int num ;
     	vic_init();
	read_vic_reg();
	delay (10);
	srand((unsigned int)time(0));
    
    	for(i= 0;i < 5;++i){
       num = rand() % 50 + 1;    
        }
  
	while(1);

	return 1;
}
编译的错误如下：
/home/hongze/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/cccEBrjF.o: in function `.L11':
main.c:(.text+0x34): undefined reference to `time'
/home/hongze/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: main.c:(.text+0x48): undefined reference to `srand'
/home/hongze/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: main.c:(.text+0x58): undefined reference to `rand'
collect2: error: ld returned 1 exit status

jim-wilson, 2020-10-29:
Did you link with the C library?  You didn't show the compile command.  Newlib does have the rand function so it should be there.  time is problematic, as it requires access to a real time clock for which there is no RISC-V standard, so you need a board specific package for your target that knows where the clock is, or you try to fake it, for instance, by using rdcycle and dividing by the clock rate mentioned in the docs.  Note that there is a rdtime instruction, but rocket chip traps on rdtime and assumes you have a trap handler in place that will access the real time clock via memory mapped I/O.  So rdtime is not very useful on rocket chip and similar cores.

hqw-lantian, 2020-10-29:
> Did you link with the C library? You didn't show the compile command. Newlib does have the rand function so it should be there. time is problematic, as it requires access to a real time clock for [...]

hqw-lantian, 2020-10-29:
> Did you link with the C library? You didn't show the compile command. Newlib does have the rand function so it should be there. time is problematic, as it requires access to a real time clock for [...]

Thanks Jim. My compilation command is like this:
riscv64-unknown-elf-gcc -march=rv64g -mabi=lp64 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -T ./link.ld  ./startup.S nvic.c timer.c main.c gpio.c -o boom.o
This is the first time I call the function library, and the previous compilations are for functions written by myself.
@jim-wilson

closed 2020-11-09 by hqw-lantian