riscv-collab/riscv-gnu-toolchain #756: How to use soft-float with nostdlib

Phantom1003, 2020-10-30:
I want to compile a bare metal coremark with -march=rv64ima -mabi=lp64, the coremark is from [riscv-boom](https://github.com/riscv-boom/riscv-coremark). This repo will generate a pk version, and a bare metal version
```
riscv64-unknown-elf-gcc -v                
Configured with: --with-abi=lp64 --with-arch=rv64ima 
```
It works well for the pk version. But for the bare metal version I also set flags `-fno-common -nostdlib -nostartfiles`, then I get:
```
riscv64-unknown-elf-gcc -march=rv64imac -mabi=lp64 -O2 -mcmodel=medany -static -std=gnu99 -fno-common -nostdlib -nostartfiles -lm -lgcc -T ../riscv64-baremetal/link.ld -I../riscv64-baremetal -I. -DFLAGS_STR=\""-march=rv64imac -mabi=lp64 -O2 -mcmodel=medany -static -std=gnu99 -fno-common -nostdlib -nostartfiles -lm -lgcc -T ../riscv64-baremetal/link.ld   "\" -DITERATIONS=0  core_list_join.c core_main.c core_matrix.c core_state.c core_util.c ../riscv64-baremetal/core_portme.c ../riscv64-baremetal/syscalls.c ../riscv64-baremetal/crt.S -o ./coremark.bare.riscv 

... undefined reference to `__ltdf2&apos;
... undefined reference to `__gtdf2&apos;
... undefined reference to `__floatunsidf&apos;
... undefined reference to `__floatunsidf&apos;
... undefined reference to `__divdf3&apos;
... undefined reference to `__floatundidf&apos;
... undefined reference to `__divdf3&apos;
```
I gest it because the `nostdlib`, so I remove it, but I got:
```
riscv64-unknown-elf-gcc -march=rv64imac -mabi=lp64 -O2 -mcmodel=medany -static -std=gnu99 -fno-common  -nostartfiles -lm -lgcc -T ../riscv64-baremetal/link.ld -I../riscv64-baremetal -I. -DFLAGS_STR=\""-march=rv64imac -mabi=lp64 -O2 -mcmodel=medany -static -std=gnu99 -fno-common  -nostartfiles -lm -lgcc -T ../riscv64-baremetal/link.ld   "\" -DITERATIONS=0  core_list_join.c core_main.c core_matrix.c core_state.c core_util.c ../riscv64-baremetal/core_portme.c ../riscv64-baremetal/syscalls.c ../riscv64-baremetal/crt.S -o ./coremark.bare.riscv 
.../libgcc.a(divdf3.o): in function `__divdf3':
.../libgcc/soft-fp/divdf3.c:46:(.text+0xb0): relocation truncated to fit: R_RISCV_HI20 against `.L10'
.../libgcc.a(_clzsi2.o): in function `__clzdi2':
.../libgcc2.c:710:(.text+0x18): relocation truncated to fit: R_RISCV_HI20 against symbol `__clz_tab' defined in .rodata section in .../libgcc.a(_clz.o)
```
But I do set `-mcmodel=medany` ... What caused these errors ? Thanks

jim-wilson, 2020-10-30:
The ltdf2 etc functions are soft-float math helper functions from libgcc.

When you use -mcmodel=medany, any C files on the gcc command line will be compiled by that option.  However, precompiled libraries that you are linking with, like libgcc, will have been compiled by the default options, which appears to be medlow.

To fix this, you need to use --with-cmodel=medany when configuring, which will ensure that all of the precompiled libraries like libgcc and libc are compiled medany.  medany code can be linked into medlow code, so it is safe to use medany by default.  This does mean that you will need to use -mcmodel=medlow if you actually want medlow code though, as medany will be the compiler default.

Phantom1003, 2020-10-30:
Thank you, add --with-cmodel=medany works!
One more question, should I remove nonstdlib for bare metal programs? It seems soft-float is a part of stdlib.

jim-wilson, 2020-10-30:
Depends on what you are doing.  The default port of newlib assumes it will run on a simulator and uses ecall to call into the simulator for things like stdio.  That won't work on bare metal.  You need board support package for your target for that to work.  A BSP probably requires use of -nostdlib to disable newlib, and then you need to add -lgcc back in because that one is still needed.  You can write your own BSP if you don't need anything fancy.  There is no standard rom monitor interface for RISC-V so no standard BSP.  There is a semihosting spec, that one can work on bare metal provided that you have something talking to the board like a debugger to intercept the semihosting calls.  There is support for this in Liviu's GNU MCU Eclipse package.  I know patches have been submitted to qemu but don't track qemu so I don't know if they have gone in.  I know that there were plans to add this to the SiFive Freedom Studio but haven't checked so don't know if it is there yet.  I don't do any bare metal programming so I'm not sure of the status of everything.
https://github.com/riscv/riscv-semihosting-spec
https://eclipse-embed-cdt.github.io/

closed 2020-10-31 by Phantom1003