riscv-collab/riscv-gnu-toolchain #757: Is it possible to build a toolchain without standart C library?

Arfarafar, 2020-11-01:
I want to make very simple project, and for my purposes it's unnecessary to use libc/libgcc. So is it possible to exclude this modules with any configure keys before make?

kito-cheng, 2020-11-01:
That's what you want, only binutils and gcc without libgcc and newlib:
```
./configure --prefix=/opt/riscv
make stamps/build-gcc-newlib-stage1
```

Arfarafar, 2020-11-01:
Ty

Arfarafar, 2020-11-02:
ok, i've done this:

```
./configure--prefix=/opt/riscv --with-arch=rv32im --with-abi=ilp32 --disable-gdb
sudo make stamps/build-gcc-newlib-stage1
export PATH=/opt/riscv/bin:$PATH
```

and now there is a problem with my linker:

```
riscv32-unknown-elf-gcc g.c 
/opt/riscv/lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: cannot find crt0.o: No such file or directory
/opt/riscv/lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: cannot find crtbegin.o: No such file or directory
/opt/riscv/lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: cannot find -lgcc
/opt/riscv/lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: cannot find -lc
/opt/riscv/lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: cannot find -lgloss
collect2: error: ld returned 1 exit status
```
what should i do?

kito-cheng, 2020-11-02:
Hmmmm, those file you need to provide if you don't any libgcc/libc, `-nostdlib` could tell gcc don't link those stuffs.

closed 2020-11-02 by Arfarafar