riscv-collab/riscv-gnu-toolchain #758: Undefined reference to memset without memset call

DurandA, 2020-11-02:
I am compiling my own bootloader for a PicoRV32 soft CPU with the `-nodefaultlibs` flag. When I enable LTO, the linker throws ```undefined reference to `memset'``` while there is no _memset()_ call.

Even more surprising, after playing with the function that supposedly use _memset()_, the compilation finishes without error if I remove a statement that modify some data in place.

Do you have any guess of what could cause this? Should I try to make a minimal reproducible example?

aswaterman, 2020-11-02:
This is normal GCC behavior, not specific to RISC-V.  If you compile the same function with `-O3` for x86-64, you'll see that it also turns into a `memset` call:

```#include <stddef.h>
void foo(char* a, char c, size_t n)
{
  for (size_t i = 0; i < n; i++)
    a[i] = c;
}
```

You can suppress the behavior with `-fno-builtin-memset`, or to suppress it for all standard C functions, `-fno-builtin`.

DurandA, 2020-11-02:
> You can suppress the behavior with `-fno-builtin-memset`, or to suppress it for all standard C functions, `-fno-builtin`.

Everything is already compiled/linked with the `-fno-builtin` flag.

aswaterman, 2020-11-02:
Not sure, but it would be trivial to provide a `memset` implementation.

DurandA, 2020-11-02:
Should it be included in a special way?

Providing my own _memset_ throws the following error:
```
/home/duranda/riscv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: libc.o (symbol from plugin): in function `_ctype':
(.text+0x0): multiple definition of `memset'; memset.o (symbol from plugin):(.text+0x0): first defined here
```

If I add `__attribute__((weak))` to the _memset_ definition, then the ```undefined reference to `memset'``` happens again.

aswaterman, 2020-11-02:
If you're linking against libc, you shouldn't need your own memset function.

DurandA, 2020-11-02:
I am linking against a [minimal libc embedded implementation](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/libbase/libc.c). That's why I couldn't redefine my own _memset_.

I am able to make a call to _memset()_ in the file that causes the issue and it compiles (if I remove the statement that is turned into a _memset_ by GCC). However, GCC doesn't want to use this implementation when optimizing.

Thank you for your time.

DurandA, 2020-11-02:
Do you know what specific optimization inserts the _memset_ call? I tried compiling with `-fno-tree-loop-distribute-patterns` but it didn't remove the _memset_ call.

kito-cheng, 2020-11-02:
Oh, I guess I know what happened, I hit similar issue servers years ago and tried to fixed that, but it turns out I misunderstanding the meaning of `-fno-builtin`[1], `-fno-builtin` not guaranteed GCC won't call memcpy, memmove or memset implicitly.

I guess the simplest solution is, DO NOT compile your `libc.c` with -flto, or in another word, compile libc.c with `-fno-lto`.

That's my guess about what happen, I don't have know how to reproduce what you see, so it might incorrect, 
- During the first phase of LTO, LTO will collect any symbol you used in program
- And then ask linker to provide those files, and discard any unused symbol.
- Then read those files into GCC and optimize again, in this moment gcc using some built-in function to optimize or code gen, but it not pull-in before.
- The symbol reference is created at LTO stage, which is too late pull in any symbol in current GCC LTO flow, and in this case, memset is discard in earlier stage...

So you might have question about why compile `libc.c` with `-fno-lto` will work? because if it didn't involved into LTO flow, which mean it won't be discarded in the LTO flow.

Some sample program to show the gcc will call memset even you compile with `-fno-builtin`, aarch64 gcc and riscv gcc will generate a function call to memset.
```
// $ riscv64-unknown-elf-gcc x.c -o - -O3  -S -fno-builtin
struct bar {
    int a[100];
};

struct bar y;

void foo(){
  struct bar x = {{0}};
  y = x;
}
```

Here is the corresponding [gcc source code](https://github.com/riscv/riscv-gcc/blob/riscv-gcc-10.2.0/gcc/expr.c#L3143)[2] for this case.

[1] https://gcc.gnu.org/pipermail/gcc-patches/2014-August/397382.html
[2] https://github.com/riscv/riscv-gcc/blob/riscv-gcc-10.2.0/gcc/expr.c#L3143

DurandA, 2020-11-02:
Thank you for the extensive explanation, that was really helpful. Thanks to both @kito-cheng and @aswaterman as this might enable us to enable LTO as default in LiteX BIOS and firmware(s) (see https://github.com/enjoy-digital/litex/issues/682).

> DO NOT compile your `libc.c` with -flto, or in another word, compile libc.c with `-fno-lto`.

Awesome, that works.

@kito-cheng Can you copy-paste your excellent answer to my [Stack Overflow question](https://stackoverflow.com/questions/64648300/how-to-prevent-gcc-from-inserting-memset-during-link-time-optimization)? I think it really deserves some exposure.

Finally, do you know what other libc functions might be inserted besides `memset` during optimization? I want to fine-tune the linking so only necessary parts of libc.c are compiled with `-fno-lto`.

kito-cheng, 2020-11-05:
>Finally, do you know what other libc functions might be inserted besides memset during optimization? I want to fine-tune the linking so only necessary parts of libc.c are compiled with -fno-lto.

memcpy, memmove, memset and abort function, I guess, but I didn't have a full list on that :P

closed 2021-03-04 by lazyparser