riscv-collab/riscv-gnu-toolchain #759: Question about how to use rvv intrinsic function

fanghuaqi, 2020-11-03:
Hi there, I found there is a branch called https://github.com/riscv/riscv-gnu-toolchain/tree/rvv-intrinsic, is this branch used for generating the rvv-intrinsic function for vector unit? If yes, how can I use it? Is there any guidance I can found?

Thanks
Huaqi

jim-wilson, 2020-11-03:
The rvv intrinsic spec can be found here.
https://github.com/riscv/rvv-intrinsic-doc
And this includes some examples showing how to use it.

fanghuaqi, 2020-11-03:
Thanks @jim-wilson,  I have already found that spec.

But I want to confirm that if I have to use this intrinisc functions defined by riscv-gcc, I need to following these steps:

* Pass extra `v` to gcc `-march=` option, eg. for rv64imacv, I need to pass `-march=rv64imacv -mabi=lp64`
* Include the `riscv_vector.h` header file, if I want to use the intrinsic functions defined by compiler

   https://github.com/riscv/rvv-intrinsic-doc/blob/master/rvv_sgemm.c#L1

* Use the `rvv_intrinsic` branch to build riscv gnu toolchain?

Is above correct?

BR
Huaqi

jim-wilson, 2020-11-03:
Yes, that looks right.  You might want to use --with-arch=rv64gcv_zfh if you want the FP16 support.

The announcement on sw-dev gives some info about how to use the toolchain.
https://groups.google.com/a/groups.riscv.org/g/sw-dev/c/Krhw8--wmi4/m/-3IPvT7JCgAJ

fanghuaqi, 2020-11-03:
Thank you very much, it really helps.

closed 2020-11-03 by fanghuaqi