riscv-collab/riscv-gnu-toolchain #760: Directory /usr/riscv64-linux-gnu/sys-root is empty

PaoloFR, 2020-11-03:
Installed package gcc-riscv64-linux-gnu.x86_64 on Fedora 33 using standard dnf command.

The directory /usr/riscv64-linux-gnu/sys-root that should conatin the usr/include/* files/directories is empty.

I have also installed package binutils-riscv64-linux-gnu.x86_64 and the qemu package

Am I doing anything wrong??

PaoloFR, 2020-11-03:
One more thing.  THe package installed version is: 10.2.1-3.fc33

tekkamanninja, 2021-03-04:
H

> Installed package gcc-riscv64-linux-gnu.x86_64 on Fedora 33 using standard dnf command.
>
> The directory /usr/riscv64-linux-gnu/sys-root that should conatin the usr/include/* files/directories is [...]

Hi @PaoloFR , mine is empty too. So you did nothing wrong. it should be empty now. 
would you let me know the header files you are looking for ? maybe they are in other directories, or in other packages ??

TommyMurphyTM1234, 2022-09-29:
No update/activity in over a year and no clarification of if/why the empty directory is a problem.

closed 2022-09-29 by TommyMurphyTM1234