riscv-collab/riscv-gnu-toolchain #764: RV32EMC Support

rahulraveendran15-coder, 2020-11-04:
Hi,

Is it possible to build the tool chain with rv32emc? I need rv32emc instruction in one of my project. How can I build the tool chain for rv32emc?

jim-wilson, 2020-11-04:
When configuring, use "--with-arch=rv32emc --with-abi=ilp32e" to get rv32emc/ilp32e as the default.  What you suggested probably works too, as the abi should default from the arch.  This will only work for an embedded elf toolchain.  Linux requires the A (atomic) extension.

rahulraveendran15-coder, 2020-11-04:
This will only work for an embedded elf toolchain. Linux requires the A (atomic) extension. >> I didnt get this point.
I am using Centos 7 and how to add A (atomic) extension? wit hwhat I need to add A extension?

jim-wilson, 2020-11-04:
The host is not relevant.  If you are building a riscv32-elf toolchain, then rv32emc is fine.  If you are building a riscv32-linux toolchain, then rv32emc won't work because the A extension is missing.  Also, we probably haven't done a glibc port to rv32e either.  Anyways, you didn't mention the target, so I was just mentioning for completeness that this won't work for a risc-v linux target.

rahulraveendran15-coder, 2020-11-04:
Sorry to ask I am new to this setup,
 may I know what is the difference between riscv32-elf toolchain and riscv32-linux toolchain? Which one should I use?

I am using the below command after cloning the riscv-gnu-toolchain repository

Installation (Linux)
To build the Linux cross-compiler, pick an install path. If you choose, say, /opt/riscv, then add /opt/riscv/bin to your PATH now. Then, simply run the following command:

./configure --prefix=/opt/riscv
make linux

If the above one which I build corresponds to riscv32-linux toolchain then how can I add A extension for linux tool chain?
Can I do something like rv32emac or rv32emca? 
 >>  Also, we probably haven't done a glibc port to rv32e either :  with linux tool chain I cant use rv32emc  in no ways?

jim-wilson, 2020-11-04:
A riscv32-elf toolchain is intended for embedded work, and uses newlib.  A riscv32-linux toolchain is intended for linux work, and uses glibc.  The embedded elf toolchain is simpler to build and use.  The linux toolchain is required if you want to build linux applications, i.e. programs that need linux system calls.  If you just want to compile small test programs, or embedded applications, then the embedded elf toolchain should be all that you need.

Linux as an operating system requires certain hardware features to function, like the atomic extension.  You can just add A to the arch, so rv32emac would be the correct order.

However, there is a bigger problem here that we don't have a linux glibc port for rv32e.  This would require changing a number of hand written assembly routines in glibc, and may also require simulator changes.  This is likely a medium size project that will require understanding of how glibc and the kernel work.  This is probably not worth the effort.

We do have a newlib port for rv32e.  It should just work as is.

rahulraveendran15-coder, 2020-11-06:
Thank you very much for explaining, actually I am doing my thesis work on RISCV and I am new to this.
I have confirmed that I will be using riscv32-elf tool chain for my thesis work, but one question is that will this support elf tool chain support bit manipulation ?
like RV32EMCB and RV32IMCB.
In Ibex github its mentioning that we need to take care that tool chain supports bit manipulation. SO I am worried about this whether the elf tool chain will support bit manipulation.

jim-wilson, 2020-11-06:
The B extension is still in draft form and still changing in incompatible ways.  The master branch of riscv-gnu-toolchain does not have B support.  We do not have a B extgension branch for riscv-gnu-toolchain as far as I know.  We do have a branch in the riscv-gcc subdir riscv-gcc-10.2.0-rvb that has B support, and this is an active development branch so incompatible changes are possible.  Likewise there is a B extension branch in riscv-binutils-gdb called riscv-binutils-2.35-rvb which is subject to change at any time.  So you can clone riscv-gnu-toolchain and then update the riscv-binutils and riscv-gcc subdirs to use the B extension branches.

rahulraveendran15-coder, 2020-11-09:
Hi,
the OVPsim simulator wont support ilp32e,  is it possible to change the ABI of elf tool chain with another API, with architecture same as RV32EMC?

jim-wilson, 2020-11-09:
No.  ilp32e is the only currently supported ABI that will work with rv32e.  The default abi ilp32 uses 8 registers for arguments, a0 to a7, where a6 is x16 and a7 is x17.  rv32e does not have a6 and a7, so you must use an alternative ABi ilp32e, that only uses 6 registers for arguments.  Normally this shouldn't affect the simulator.  However, the simulator support assumes that syscall numbers are placed in a7, so ilp32e uses an alternative scheme that places syscall numbers in t0.  If you need to make syscalls and have a simulator emulate them, then you need simulator support for rv32e.  If you don't need simulator syscall support, then ilp32e should work fine on any simulator.

rahulraveendran15-coder, 2020-11-13:
(1) Can I use the same tool chain as RV32EMC and RV32IMC, by passing -march=RV32IMC or RV32EMC while compiling?
The command I used to build tool chain is  `./configure --prefix=/opt/riscv --with-arch=rv32emc`

(2) how can I rebuild the platform from RV32EMC  to RV32IMC? if (1) is not possible Can I again rebuild the tool chain from the same clone which I created with the command "`/configure --prefix=/opt/riscv --with-arch=rv32imc`" ,can I switch between ISA  like this from same git repository or do I need to clone again a new gnu-tool chain repo and configure/build new tool chain with rv32imc and add Path of those?

(3) When I am building the tool chain with RV32EMC  and on compilation of a c file with "-march=rv32imc" works well at the same time building the tool chain with RV32IMC and runnign the c fiel with "-march=rv32emc --mabi=ilp32e" doesnt works, what can be the reason for this?

jim-wilson, 2020-11-14:
The compiler will support any reasonable arch/abi command.  But if you want to link with precompiled libraries, then you need to enable multilibs, and the multilib list must include the arch/abi choices you want.  The default list only has a few and may not have what you need.  It is easier to just build two compilers.  Or you could try the SiFive toolchain release as it has more choices than the default FSF sources.

We recommend that you do not build the toolchain in the source tree.  E.g. clone riscv-gnu-toolchain, then mkdir X-build (or whatever), cd X-build, and run ../riscv-gnu-toolchain/configure ...  If you do builds this way, then you can build multiple toolchains from the same source tree.  It is also easier to clean up after a mistake, as you can just rm -rf the build tree.

rv32imc/ilp32 code is not run-time compatible with rv32emc/ilp32e code.  They are are different ABIs.  You can't link the code together.  Otherwise, it isn't clear what your problem is.

rahulraveendran15-coder, 2020-11-15:
@jim-wilson 
My need is I want to run or compile my  c program with  RV32EMC and some times with RV32IMC.
So do I need to take two clones of riscv  tool chain and build with --with-arch=rv32emc in clone 1 for RV32EMC and in another 2nd clone with --with-arch=rv32imc for RV32IMC? Or can I create a build-imc and build-emc directory as you said  before to create two different(arch) tool chain support?

Concern-Does a tool chain built with RV32EMC is compatible with RV32IMC(Below (1) I am observing don't know this is intended or not)?

(1) When I am building the tool chain with RV32EMC and on compilation of a c file with "-march=rv32imc" it works well means the executable file i am able to create when I built a tool chain with RV32EMC and run `riscv-unknown-elf-gcc -o hello hello_world.c -march=rv32imc.

When I am building the tool chain with RV32IMC and on running a c file with "-march=rv32emc" it doesnt works means the executable file i am able to create when I built a tool chain with RV32IMC and run `riscv-unknown-elf-gcc -o hello hello_world.c -march=rv32emc. This is causing error.
`

jim-wilson, 2020-11-15:
You can clone one copy of riscv-gnu-toolchain, and then have two build dirs one for rv32emc and one for rv32imc.

rv32emc code is not compatible with rv32imc code.  rc32imc uses 8 arg registers, rv32emc uses 6 arg registers.  rc32imc uses a7 for system call numbers.  rv32emc uses t0 for system call numbers.

You didn't mention exactly what error you get when you try to mix rv32emc and rv32imc code, but this isn't expected to work.

lazyparser, 2021-03-04:
@rahulraveendran15-coder Hi, is this issue resolved? Can we close it safely?

rahulraveendran15-coder, 2021-03-04:
Sure, Thank you very much for the support, you can close this issue

closed 2021-03-04 by rahulraveendran15-coder