riscv-collab/riscv-gnu-toolchain #766: can't link hard-float modules  with soft-float modules

hqw-lantian, 2020-11-09:
My software code is as follows

#include "reg.h"
#include "pub_def.h"
#include "nvic.h"
#include "encoding.h"
#include "gpio.h"
#include "pub.h"

#include <stdio.h>
#include <stdlib.h>
#include <time.h>

int main(void)
{
	int i = 0;
 	int num ;
     	vic_init();
	read_vic_reg();
	delay (10);
	srand((unsigned int)time(0));
    
    	for(i= 0;i < 5;++i){
       num = rand() % 50 + 1;     
        }
  
	while(1);

	return 1;
}

The compilation command is as follows

riscv64-unknown-elf-gcc -march=rv64g -mabi=lp64 -static -mcmodel=medany -fvisibility=hidden  -nostartfiles -T ./link.ld ./startup.S nvic.c timer.c main.c gpio.c -o boom.o

The compilation error is roughly as follows

/riscv64-unknown-elf/bin/ld: /home/1234/0303_2/riscv/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/libgcc.a(_clz.o): can't link hard-float modules with soft-float modules
/riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /home/1234/0303_2/riscv/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/libgcc.a(_clz.o)
/riscv64-unknown-elf/lib/libc.a(lib_a-printf.o): In function `printf':
printf.c:(.text+0x26): relocation truncated to fit: R_RISCV_HI20 against symbol `_impure_ptr' defined in .sdata section in /home/1234/0303_2/riscv/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)
/home/1234/0303_2/riscv/bin/../lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: final link failed: Symbol needs debug section which does not exist
collect2: error: ld returned 1 exit status

jim-wilson, 2020-11-09:
The "-march=rv64g -mabi=lp64" options on the comand line will affect your C code, but won't affect precompiled libraries, which will have been compiled using the defaults which are probably rv64gc/lp64d.  There are two ways to fix this.  You can build a toolchain with the right default, e.g. use --with-arch=rv64g --with-abi=lp64 when configuring the toolchain.  Or you can enable the multilib support with --enable-multilib that will build multiple copies of the libraries with different arch/abi options.  However, there is a limited set of them and rv64g/lp64 is not one of them.  One of them is rv64imac/lp64 if that works for you.  Otherwise you can try changing the set of default multilibs, but using the right --with-arch/--with-abi options is the simpler solution.

For the R_RISCV_HI20 relocation truncated error, there are multiple ways to hit this, but I notice that you are using -mcmodel=medany.  Like the above, this will apply to your C code, but not to the precompiled libraries which would have been compiled with the default.  If the toolchain default is medlow this could explain why you hit this error.  You can fix this by using --with-cmodel=medany when configuring the toolchain to get medany precompiled libraries.

The last error is a cascading error, and a known bug in the gcc-7.2.  This compiler version is known to be broken and should not be used for any purpose.  Anything gcc-7.3 or later should be OK, though currently we are on gcc-10.

hqw-lantian, 2020-11-11:
Thank you for your serious answer@jim-wilson

lazyparser, 2021-03-04:
@hqw-lantian Hi, is this issue fixed? can we close this issue safely?

jim-wilson, 2021-03-04:
Since these aren't paying customers, I would suggest that if we think a problem has been answered, and it has been more than a few weeks since the original poster responded, that we should just close it.  The original poster can object if they think it isn't fixed yet.  I think this approach is less work for us, since few are likely to object to closing issues for answered questions.

lazyparser, 2021-03-05:
> Since these aren't paying customers, I would suggest that if we think a problem has been answered, and it has been more than a few weeks since the original poster responded, that we should just [...]

Fair enough. I will work through the open issues and close all issues seems answered/solved. If I closed it wrong, anyone can just reopen it.

closed 2021-03-05 by lazyparser