riscv-collab/riscv-gnu-toolchain #767: link-time optimization with static linked libraries

mmahdigit, 2020-11-10:


kito-cheng, 2020-11-10:
You need riscv32-unknown-elf-gcc-ar and riscv32-unknown-elf-gcc-ranlib, you could try to add `-DCMAKE_RANLIB=/opt/riscv/bin/riscv32-unknown-elf-gcc-ranlib -DCMAKE_AR=/opt/riscv/bin/riscv32-unknown-elf-gcc-ar` to cmake command.

mmahdigit, 2020-11-10:
> You need riscv32-unknown-elf-gcc-ar and riscv32-unknown-elf-gcc-ranlib, you could try to add `-DCMAKE_RANLIB=/opt/riscv/bin/riscv32-unknown-elf-gcc-ranlib -DCMAKE_AR=/opt/riscv/bin/riscv32-unknown-elf-gcc-ar` to cmake command.

Yes, as I mentioned, using `riscv32-unknown-elf-gcc-ar` linking works, but shouldn't we have the option to enable lto for `riscv32-unknown-elf-ar` so that it links naturally?

kito-cheng, 2020-11-11:
> Yes, as I mentioned, using riscv32-unknown-elf-gcc-ar linking works, but shouldn't we have the option to enable lto for riscv32-unknown-elf-ar so that it links naturally?

Yeah, but it might out of scope of RISC-V toolchain, it more like GNU toolchain issue, I would suggest you raise discussion on upstream instead.

TommyMurphyTM1234, 2022-09-27:
See @kito-cheng's previous comment.

closed 2022-09-27 by TommyMurphyTM1234