riscv-collab/riscv-gnu-toolchain #768: LTOed libgcc and builtins

stefanct, 2020-11-10:
Hi,
I am trying to include _everything_ at the LTO stage in my build because I have a GCC plugin that instruments all functions with (HW) CFI instructions and this would increase the coverage (to a ridiculous level but bear with me please). I am working on the PULP platform and have used their fork previously but my problem equally happens with current master of this repository as well.

To that end, I have added the necessary flags (and maybe more than that) to `Makefile.in` to build even libgcc with LTO. See my [branch](https://github.com/stefanct/riscv-gnu-toolchain/commits/lto) for details, in particular 9a89389 (some other patches like the ones on .gitignore, parallel build might even be interesting for upstreaming - let me know if I should create a dedicated PR).

My problem now is that when I try to build an application that uses `__builtin_*` functions the linker step fails. For example, when I use `__builtin_clz` in the code the linker prints `undefined reference to '__clzsi2'`. This is quite surprising, because this function is included in the libgcc.a I am linking with, and when I use that directly, i.e. if I change `__builtin_clz` to `__clzsi2` in the code it links just fine. When I look at the disassembly of the object file that contains the builtin call it looks to me like it inlined everything but this does not (directly) explain the missing reference - rather the opposite :) However, maybe this is similar to #758?

I have attached a makefile that creates the necessary files (kind of broken "CRT", linker script (stolen from pulp), and `missing.c` with the respective calls) to show the problem.
Just rename to `makefile` and execute `make` with the LTO-enabled `riscv32-unknown-elf-gcc` from my branch in your path to see the undefined reference. If you either enable the non-`__builtin_*` calls in `missing.c` or disable LTO linking (1. line of makefile), linking works just fine.

What's happening?

BTW: executing `make ARCH=anything` will build the example with the host CC and RT (expecting something x86) to show everything works as intended with ordinary toolchains.

[makefile.txt](https://github.com/riscv/riscv-gnu-toolchain/files/5518670/makefile.txt)

ilg-ul, 2020-11-10:
Not directly related to your questions, but related to _LTO everything_.

Some time ago I experimented with the xPack toolchains with LTO in newlib and libstdc++. The build was ok, and the toolchain was functional, but all object files were roughly 10 times larger, which, with multilibs, resulted in a .tar.gz of more than 3 GB.

If you build your toolchain for local purposes it should be ok, but if you plan to distribute it... I don't know, the size might be a limiting factor.

stefanct, 2020-11-10:
Thanks for your considerations. My gcc build/install directory is "only" 1.4GB uncompressed... but it's not any issue anyway since this is just a research prototype.

ilg-ul, 2020-11-10:
>  is "only" 1.4GB

If you build newlib + newlib-nano in all multilib variants it might get larger.

kito-cheng, 2020-11-11:
> My problem now is that when I try to build an application that uses __builtin_* functions the linker step fails. For example, when I use __builtin_clz in the code the linker prints undefined [...]

Yes, that's same issue, so I would suggest you don't compile libc, libgcc and crt* files with LTO.

TommyMurphyTM1234, 2022-09-29:
The issue here is arguably inappropriate use of LTO and not any specific problem with the RISC-V GCC toolchain itself.

closed 2022-09-29 by TommyMurphyTM1234