riscv-collab/riscv-gnu-toolchain #770: Facing issue while generating hex file

Prarthi1703, 2020-11-12:
I am generating a hex file using the below command, but it is not generating the hex file properly and elf,objdump and readelf are getting generated.

$RISCV_BIN/riscv64-unknown-elf-gcc -g hello-world.c -o hello-world.o.
$RISCV_BIN/riscv32-unknown-elf-objcopy  -S -O ihex fibonacci.o fibonacci.hex

jim-wilson, 2020-11-12:
Why do you think that the hex file was not generated properly?  I can't help when I don't know what is wrong.
]
If hex file support is broken, that isn't a RISC-V bug, and should be reported upstream instead of here.

SiFive recommends use of the elf2hex program to generate hex files for verilog simulators.
https://github.com/sifive/elf2hex

Prarthi1703, 2020-11-13:
Hi,

While generating hex file, elf file and readelf is not getting generated.

I am using the below command to generate a hex file.

$RISCV_BIN/riscv64-unknown-elf-gcc -g hello-world.c -o hello-world.o.
$RISCV_BIN/riscv32-unknown-elf-objcopy  -S -O ihex fibonacci.o fibonacci.hex

Thank
Thanks & Regards
Prarthi Bhatt

jim-wilson, 2020-11-13:
It still isn't clear to me what the problem is.  Can you show me an actual error?

The objcopy command only creates one output file, the hex file.  It will not create an elf output file.  readelf is a program which you aren't trying to use, so not clear why you are referring to readelf.

Prarthi1703, 2020-11-13:
Hi,

I have attached hex file below.

[image]

Prarthi1703, 2020-11-13:
Hi,

I have install riscv-gnu-toolchain using https://github.com/riscv/riscv-gnu-toolchain repository and I have install riscv 32 bit library. 

After riscv-gnu-toolchain, I am trying to generate hex file. Below I have attached hex file that is getting generate by toolchain.

[image]

Prarthi1703, 2020-11-13:
I also need elf file to run uvm testcase.

jim-wilson, 2020-11-13:
That looks like a valid hex file, so it isn't clear what the problem is.

Perhaps the confusion is with how to run the compiler?  It appears that you are using gcc -c to create an object file, and then passing the object file through objdump to create a hex file.  You haven't said what you are trying to do with this hex file, but if you are trying to pass it to a simulator I would not expect that to work.  You can't execute object files in general.  You need to produce a fully linked executable first, linked against start files and libraries, then create a hex file from that, and then pass that to a simulator.

MarcKarasek, 2020-11-16:
I would also suggest using this:  https://github.com/sifive/freedom-elf2hex.git

As i had problems generating hex files with the other elf2hex program.

TommyMurphyTM1234, 2022-09-27:
No update/activity in over a year and the original question seems to be about how to use the tools as opposed to reporting an issue with the RISC-V GCC toolchain itself.

closed 2022-09-27 by TommyMurphyTM1234