riscv-collab/riscv-gnu-toolchain #771: Facing error in enabling the profiler for elf toolchain.

subhajit26, 2020-11-13:
Hi,
      I have build an elf toolchain and I am trying to enable the profiler with the command "riscv32-unknown-elf-gcc -static -pg fibonacci.c -o fibonacci" . 

On executing the above command it gives me an error:

fibonacci.c:(.text+0x12): undefined reference to `_mcount'
/home/sbhuinya/riscv-gnu-toolchain/lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/ccP9AZOz.o: in function `main':
fibonacci.c:(.text+0x64): undefined reference to `_mcount'
collect2: error: ld returned 1 exit status.

Can you please help me out to resolve the issue as  I need to run the riscv32-unknown-elf-gprof  since its part of my thesis.

-

jim-wilson, 2020-11-14:
gprof style profiling requires timer interrupts which implies OS support.  On unix systems it uses the profil system call.  Try "man profil" on any unix system.  It is probably infeasible to implement this for a bare metal embedded system.

gprof without the call graph, i.e. only the flat profile should be possible, but will require writing library support, as no one has done this for RISC-V yet.  You can find examples of mcount support in riscv-newlib/libgloss for other targets.  You can probably copy one into the RISC-V dir and edit it to work for RISC-V, like maybe the one in arc and rx dirs that looks like the same mcount implementation.  You can try using a CSR like mcycle/mcycleh for timing info, though I think the existing ones only count the number of calls.

The next problem will be dumping the info, since they expect the program to exit and then dump info to a file at exit, which may not be possible on a bare metal system.  The embedded app may not exit, and may not have a filesystem.  But there are workarounds such as using gdb to read the data from the target memory.  Or maybe you can set up a semihosting environment so the target can write to files.  You may need to add a hook to dump the data on command instead of waiting for program exit.  This will probably require writing some more code.

TommyMurphyTM1234, 2022-09-29:
@jim-wilson has explained why gprof is not automatically available for the bare metal toolchain and why additional, possibly custom and board specific, support is needed for it to work.

closed 2022-09-29 by TommyMurphyTM1234