riscv-collab/riscv-gnu-toolchain #772: Error: unrecognized opcode `c.jal 1b'

MarcKarasek, 2020-11-13:
Compiling the compliance tests,  it fails on cjal-01.S with Error: unrecognized opcode `c.jal 1b'

All the other compressed test cases pass compilation.

jim-wilson, 2020-11-14:
You didn't include the cjal-01.S file, or tell us where to find it.  I would guess that it is in riscv-compliance, but I see no such file in there.

MarcKarasek, 2020-11-14:
It is in a dev branch on another fork that is going to be PR soon.

The source file will not show you anything though as it uses macros to
generate the test and the source.
I can add the source file plus the macros file to the issue in jira.  This
will give show you what source is being generated at compile time.

Marc

MarcKarasek, 2020-11-14:
Here are the two files the one is the src that is compiled and the other has the macros it uses.

[cjal-01.zip](https://github.com/riscv/riscv-gnu-toolchain/files/5540095/cjal-01.zip)

Nelson1225, 2020-11-16:
I have looked your attached quickly, and notice that the TEST_CJAL_OP in the compliance_test.h only use ".option push" to record the previous status before ".option norvc", but I don't find any place to enable the c-extension.  Therefore, I guess you have to enable the c-extension by the assembler option -march= by yourselves, or change the header file to enable the rvc by default, or use the toolchain which was built to enable the c-extension (built with --with-arch= configure option rather than the multi-lib).

MarcKarasek, 2020-11-16:
The same basic code works for other c opcodes c.jalr compiled ok.   I will
look at this tomorrow and get back to you.

God Bless..

MarcKarasek, 2020-11-16:
Yes the compiler is built with the --with-arch option

./configure --prefix=/opt/riscv64 --with-cmodel=medany --with-arch=rv64imafdc --with-abi=lp64d

[cjalr-01.zip](https://github.com/riscv/riscv-gnu-toolchain/files/5547240/cjalr-01.zip)

Here is the c.jlar source file.  It uses the same macros as the cjal file I attched previously.

jim-wilson, 2020-11-16:
The testcases use header files that aren't included and hence aren't useful.  Note that gcc --save-temps on a .S file will produce a .s file that we can use as a testcase.

But I don't see an obvious ".option rvc" anywhere.  This is required even if you have configured for an arch that includes the C extension.

MarcKarasek, 2020-11-16:
I have checked all the other test cases for c.xxx opcodes and none of them
have .option rvc set.  Yet they all compile.

The only one that throws an unknown opcode error is c.jal which also has a
c.nop in the same code test.

I will generate the .s file from the compile and attach it to the issue.

MarcKarasek, 2020-11-16:
here is the saved-temp file

jim-wilson, 2020-11-17:
I don't see a testcase in your last comment.

MarcKarasek, 2020-11-17:
[cjal-01 (2).zip](https://github.com/riscv/riscv-gnu-toolchain/files/5554675/cjal-01.2.zip)

jim-wilson, 2020-11-17:
I don't see a problem.  Except that c.jal is an rv32 C instruction only.  Maybe that is where you are getting confused?  This testcase can not be compiled for rv64 because c.jal does not exist there.

MarcKarasek, 2020-11-17:
I am using a riscv64-unknown-elf- compiler to compiler these tests.   But the tests are 32 bit.  
So I need to compile this test with the 32 bit compiler?
or is there an option to pass to the 64 bit compiler to tell it to compile with 32 bit opcodes?

Marc

MarcKarasek, 2020-11-17:
If I add -mabi=ilp32d -march=rv32imafdc to the gcc options it compiles.
Is this a correct solution?

jim-wilson, 2020-11-17:
Yes, you need to pass arch/abi options if using a riscv64-* compiler.  The ones you suggested are fine.

MarcKarasek, 2020-11-17:
ok I will add these options to the compliance test gcc options for the example target when someone is using the rv64 compiler.

closed 2020-11-17 by MarcKarasek

manasabatchu, 2023-09-06:
I am trying to adding packed SIMD instruction to riscv-gnu-toolchain I am getting this error 
 add8.c: Assembler messages:
add8.c:8: Error: unrecognized opcode `add8 a5,a5,a4'
Binutils modification
Now, binutils need to be aware of the new instruction. riscv-gnu-toolchain/riscv-binutils/include/opcode/riscv-opc.h is updated as follows:

/* Instruction opcode macros.  */
+ #define MATCH_ADD8 0x48000077
+ #define MASK_ADD8  0xfe00707f

#endif /* RISCV_ENCODING_H */
#ifdef DECLARE_INSN
+ DECLARE_INSN(add8, MATCH_ADD8, MASK_ADD8)

The related C file (riscv-gnu-toolchain/riscv-binutils/opcodes/riscv-opc.c) has to be modified as well:

/* name, xlen, isa, operands, match, mask, match_func, pinfo.  */
+ {"add8",         0, INSN_CLASS_P, "d,s,t",     MATCH_ADD8 MASK_ADD8, match_opcode, 0 },
same files I modified in gdb folders also

manasabatchu, 2023-09-06:
> I am trying to adding packed SIMD instruction to riscv-gnu-toolchain I am getting this error add8.c: Assembler messages: add8.c:8: Error: unrecognized opcode `add8 a5,a5,a4' Binutils modification [...]

manasabatchu, 2023-09-06:
I am trying to adding packed SIMD instruction to riscv-gnu-toolchain I am getting this error
add8.c: Assembler messages:
add8.c:8: Error: unrecognized opcode `add8 a5,a5,a4'
Binutils modification
Now, binutils need to be aware of the new instruction. riscv-gnu-toolchain/riscv-binutils/include/opcode/riscv-opc.h is updated as follows:

/* Instruction opcode macros. */

#define MATCH_ADD8 0x48000077
#define MASK_ADD8 0xfe00707f
#endif /* RISCV_ENCODING_H */
#ifdef DECLARE_INSN

DECLARE_INSN(add8, MATCH_ADD8, MASK_ADD8)
The related C file (riscv-gnu-toolchain/riscv-binutils/opcodes/riscv-opc.c) has to be modified as well:

/* name, xlen, isa, operands, match, mask, match_func, pinfo. */

{"add8", 0, INSN_CLASS_P, "d,s,t", MATCH_ADD8 MASK_ADD8, match_opcode, 0 },
same files I modified in gdb folders also